stereolabs/zed-sdk #229: [ZED][Object Detection] Segmentation Fault

ilkarman, 2020-03-23:
I'm running the Cuda 9.0 version of the 3.1.1 SDK (Ubuntu 16) and enabling object_detection in the [example](https://github.com/stereolabs/zed-examples/blob/master/object%20detection/python/object_detection_2d.py):

```zed.enable_object_detection(obj_param)```

gives me the following error:

```
[ZED][Object Detection] Please wait while the AI model is being optimized for your graphics card
 This operation will be run only once and may take a few minutes  -TensorRT error in yolo.cpp : bool Yolo::createYOLOEngine(nvinfer1::Data
Type, std::__cxx11::string, Int8EntropyCalibrator*) at 874.
 Done
[ZED][Object Detection] Please wait while the AI model is being optimized for your graphics card
 This operation will be run only once and may take a few minutes  /Segmentation fault (core dumped)
```

adujardin, 2020-03-23:
What is your hardware configuration, especially the GPU?

It could be that there's not enough memory to create the engine (the initial model optimization takes more memory than during runtime).

ilkarman, 2020-03-24:
> What is your hardware configuration, especially the GPU?
> 
> It could be that there's not enough memory to create the engine (the initial model optimization takes more memory than during runtime).

Thanks for the reply. I have a GTX 2080 with 12GB (Nvidia Driver 410.104). This is my main deep-learning station so it can definitely run torch/tensorflow etc. I am executing all of these within docker containers though.

Do you know if there is a way for me to trace the error since I assume segmentation fault is not the full trace? Or perhaps disable optimisation (is doing cudnn exhaustive search for the CNN algo?) or run within cudnn?

adujardin, 2020-03-24:
Your GPU definitely has enough memory to run. Unfortunately, you can't have the full trace. 

What's running here is that TensorRT is optimizing the model to your specific GPU to generates an engine. Several things are going on but basically, TensorRT tries all the algorithm to see which variant is the fastest, for a RTX 20XX it will also quantize to at least FP16, there's also network pruning and layers fusion. TensorRT is using cuDNN is the backend.
You don't have control over this, it tries by itself the best possible and automatically handles it.

A segfault at this step is usually caused by a lack of GPU memory (definitely not the case here) or sometimes a corrupted model file. This typically applied when running natively so Docker might have something to do with this (like Tensor Core support ?). 

The model file is stored in `/usr/local/zed/resources/`. The md5 checksum of objects_1.2.model should be `ebac8fb3a731022b001d44f713ef0e19`. The engine ('optimized model'), once generated is located next to the model, for instance with a name like '.objects_1.2.model_optimized-dcabl-75-10000-7605-7000'.
When running there are some temp files generated that can also be corrupted. Natively rebooting is enough to clear them, in docker I'm not sure what's the behavior to clean the /tmp folder.

To try to replicate this, could you give me the docker image used and the command you used to run it?

closed 2022-04-27 by github-actions[bot]