stereolabs/zed-sdk #231: Spatial mapping with ZED2 and ROS

stapperfend, 2020-03-29:
Hello,
we want to use the ZED2 in combination with ROS (Melodic) to create a spatial map (mesh). Unfortunately, the zed-rtabmap-example is not working with the ZED2 and also the tutorial from rtabmap does not work, because the zed-wrapper got changed and no longer contains the URDF-files (as far as I analyzed it). There it says, to launch "zed_no_tf.launch" which requires some URDF-files.
What would be the workaroung to get the ZED2 running and to create a spatial map of the environment? Is an updated Version of the zed-rtabmap-example planned?

Myzhar, 2020-03-30:
@stapperfend this is the wrong repository for ROS issues.
The problem with RTABmap has been just reported and we are working on it:
https://github.com/stereolabs/zed-ros-examples/issues/10
It will be fixed as soon as possible

stapperfend, 2020-03-30:
Didn't see that. Sorry and thank you.

closed 2020-03-30 by stapperfend