stereolabs/zed-sdk #232: How to estimate depth using captured images from ZED , without the need of camera

poornimajd, 2020-04-07:
I do not have the ZED camera,but the dataset I am working on, consists of  the images captured from ZED camera.I wanted to know if it is possible to estimate the depth from the captured images , without physically having the camera.
Any suggestions are greatly appreciated.
Thanks

obraun-sl, 2020-04-07:
Hi,
You can if it's a SVO file you have as input. 
SVO file are recorded video from a ZED Camera with a specific format compatible with SDK inputs.
Best,
OB

poornimajd, 2020-04-07:
@obraun-sl  Thanks for your quick reply. I actually have the stereo images captured from the Zed camera,but I do not have the camera with me.Is there any code to convert the stereo images to svo format and then use it for depth estimation?
Any suggestion is greatly appreciated.
Thanks

obraun-sl, 2020-04-07:
Unfortunately, the SVO file contains more than just the two images to be able to be ingested into the ZED SDK pipeline. 
With only the 2 images, I am afraid we cannot reconstruct the SVO file. 
Maybe you can ask the dataset provider if they have the original SVO file ?

poornimajd, 2020-04-07:
Thanks for the response! I will try asking the dataset provider for the SVO file.

seanmcleod, 2020-04-16:
@obraun-sl however in the streaming case isn't the ZED SDK pipeline on the receiving end only receiving image data? Or is some non-image data, like the non-image data in the SVO file, also being streamed?

In other words if you can't get hold of the SVO file could you not video encode and stream image data into the ZED SDK pipeline? Obviously with all the extra overhead of video encoding and decoding.

obraun-sl, 2020-04-16:
@seanmcleod The streaming part is quite similar to the SVO file. It also contains non-image metadata similar to the SVO file.

seanmcleod, 2020-04-16:
@obraun-sl okay so it's not just plain H264/265 video data then?

In terms of the Zed2 is the sensor data from the IMU etc. also streamed?

And so without any documentation on the non-image metadata I guess that rules out us being able to do some testing and training with synthetic image data from something like AirSim with the Zed SDK before taking the Zed camera out into the real-world for additional testing etc.?

seanmcleod, 2020-04-20:
@obraun-sl any comments regarding my query above?

obraun-sl, 2020-04-21:
The SVO and the streaming format contains more than the video data. It also contains metadata for the image and sensors data (such as imu, and more if using the ZED2). 
However, we can think about a way to interface with simulation platform like AirSim using the streaming interface.

clintlombard, 2020-04-21:
@obraun-sl This would be really useful!

closed 2022-04-27 by github-actions[bot]