stereolabs/zed-sdk #233: In SVO recording, initParameter `depth_maximum_distance` doesn't work

poorboy44, 2020-04-07:
I am recording an SVO recording using the `svo_recording.py` script. I am setting the initParameters as follows:
```
    init = sl.InitParameters()
    init.camera_resolution = sl.RESOLUTION.HD1080
    init.depth_mode = sl.DEPTH_MODE.ULTRA
    init.coordinate_units = sl.UNIT.METER
    init.depth_maximum_distance = 1 # max 1 meter
    init.coordinate_system = sl.COORDINATE_SYSTEM.RIGHT_HANDED_Y_UP # Use a right-handed Y-up coordinate system
```
However, when I play my recording back, the max depth is set to the default (20000), not the 1 meter value that I clamped it to. This is a problem for me because it causes my point cloud to be noisier than it needs to be.  This doesn't seem to be an issue in Live mode.

```
-memories/capture_zed$ python svo_recording_sn.py scotty_selfie.svo
SVO is Recording, use Ctrl-C to stop.
^C(streaming_memories) poorboy44@poorboy44-XPS-8930:~/streaming_memories/streami-memories/capture_zed$ python zed_svo_examples/read_svo.py scotty_selfie.svo 
Reading SVO file: scotty_selfie.svo
  Save the current image:     s
  Quit the video reading:     q

Do you want to display camera information? [y/n]: y

Distorsion factor of the right cam before calibration: [-0.0402921   0.00722651 -0.00034453 -0.00025147 -0.00357014].
Distorsion factor of the right cam after calibration: [0. 0. 0. 0. 0.].

Confidence threshold: 100
Depth min and max range values: 499.05194091796875, 20000.0
Resolution: 1920, 1080.
Camera FPS: 30.0
Frame count: 92.

```

Can you confirm/advise?

P-yver, 2020-04-10:
Hi,
In the sample, [here](https://github.com/stereolabs/zed-examples/blob/master/svo%20recording/recording/python/svo_recording.py#L41), we specify that we do not need the depth information.
In fact, when you record a SVO, only the images (and IMU data) are stored, changing the `depth_maximum_distance` has no impact on the recorded SVO.

You should set these parameters on the programs that read this SVO.

poorboy44, 2020-04-10:
I figured this out. I needed to set the InitParameters when I process the SVO file. I thought InitParameters only applied to the Live video recording.  When I do this, the depth of the point cloud appears properly clamped.

closed 2020-04-10 by poorboy44