stereolabs/zed-sdk #238: What'more, I wanna know how can you convert the 2D coordinate to 3D coordinate.

sennnnn, 2020-04-11:


obraun-sl, 2020-04-13:
There is an example of convert 2D pixels coordinates to 3D world coordinates here  : 
https://github.com/stereolabs/zed-examples/blob/master/tutorials/tutorial%203%20-%20depth%20sensing/cpp/main.cpp#L67

Using the ZED SDK API.

closed 2020-04-14 by sennnnn

FeiSong123, 2023-02-23:
What I want to make sure is that as long as the xy filled here is the object xy in the left or right view, I can get the 3d point of the target object, right? Or do we have to deal with xy?

Myzhar, 2023-02-23:
Hi @FeiSong123 
please do not reply to a closed issue. The GitHub issue is for reporting bugs and feature requests.
Please consider posting on the Stereolabs forum or search the Support page for a solution, like this:
https://support.stereolabs.com/hc/en-us/articles/4554115218711-How-can-I-convert-3D-world-coordinates-to-2D-image-coordinates-and-viceversa-

FeiSong123, 2023-02-23:
Sorry, I'll try it.