stereolabs/zed-sdk #239: dataset svo

omar-Fouad, 2020-04-23:
Hi
would you please tell me if i can download svo dataset for testing example
thanks

obraun-sl, 2020-04-27:
Hi. You can find some SVO samples here : 
https://www.stereolabs.com/download_327af3/SVO_samples/

We are working to add more SVOs into public access.

omar-Fouad, 2020-04-27:
Merci beaucoup

closed 2020-04-27 by omar-Fouad

jonasdv97, 2021-03-08:
@obraun-sl, I cannot access the SVO sample. I would like to test it as well.

Thanks in advance.

emeQee, 2021-04-25:
> Hi. You can find some SVO samples here :
> https://www.stereolabs.com/download_327af3/SVO_samples/
> 
> We are working to add more SVOs into public access.

The SVO Sample links died and redirect to the front page. Do you have current links? I would appreciate it very much.

obraun-sl, 2021-04-25:
While we are migrating the SVO samples to the website you can find the links below : 

SVO samples : 
ZED : https://download.stereolabs.com/assets/svo_samples/wall
ZED : https://download.stereolabs.com/assets/svo_samples/sofa
ZED : https://download.stereolabs.com/assets/svo_samples/well
ZED2 : https://download.stereolabs.com/assets/svo_samples/ZED2_Lake_H264 (*_H265 for H265 version)
ZED2 : https://download.stereolabs.com/assets/svo_samples/ZED2_Street_H264 (*_H265 for H265 version)
ZED2 : https://download.stereolabs.com/assets/svo_samples/ZED2_Runners_H264 (*_H265 for H265 version)

Mesh from SVO samples : 
https://download.stereolabs.com/assets/mesh_samples/wall
https://download.stereolabs.com/assets/mesh_samples/sofa
https://download.stereolabs.com/assets/mesh_samples/well

ymtoo, 2021-12-08:
I'm not able to read the SVO samples, e.g.,
```
>>> import pyzed.sl as sl
>>> zed = sl.Camera()
>>> input_path = "sofa.svo"
>>> init_parameters = sl.InitParameters()
>>> init_parameters.set_from_svo_file(input_path)
>>> zed = sl.Camera()
>>> err = zed.open(init_parameters)
# get stuck here
```
My setup is:
- Ubuntu 20.04.3
- ZED SDK 3.6.2