stereolabs/zed-sdk #242: [ZED SDK] Verifying archive integrity...Error in MD5 checksums

joshferns, 2020-05-01:
Hi, I've been trying to install ZED SDK 3.1, but every time I run into the issue 

> Verifying archive integrity...Error in MD5 checksums: 37065f4fd7121fcbed043f6b7830c80a is different from b73330ff110594863356ba57b2c02996

I've even tried downloading the file from a different source. there were no errors in the downloading process.

adujardin, 2020-05-01:
Hi, this usually means there was no disk space left, and the file got truncated. 
But it could also be a CDN problem on our side. Which installer did you try exactly?

joshferns, 2020-05-01:
Hi,
We're trying

> ZED_SDK_Ubuntu16_cuda10.0_v3.1.2.run

There is enough space on the drive,
I have attempted this on 2 different computers (both Ubuntu 16.04)
and even tried downloading from 2 different sources.

Initially we had an older download 
> ZED_SDK_Ubuntu16_cuda10.0_v3.0.1.run
Today was the first time we were attempting to install it, 
It had similar issues. which is why we tried getting version 3.1.2

adujardin, 2020-05-01:
I tried different installers from different places and I can't replicate it. 

You could try the direct links:
- Without CDN: https://stereolabs.sfo2.digitaloceanspaces.com/zedsdk/3.1/ZED_SDK_Ubuntu16_cuda10.0_v3.1.2.run
- CDN: https://stereolabs.sfo2.cdn.digitaloceanspaces.com/zedsdk/3.1/ZED_SDK_Ubuntu16_cuda10.0_v3.1.2.run

If it still doesn't work, let me know and I'll upload it somewhere else as a workaround (not ideal, but we'll try to identify what's going on)

joshferns, 2020-05-02:
Hi,
The one without CDN gives the checksum issue.
The link with CDN seems to work.

FENGZHANG123, 2020-05-04:
I also met this problem. And downloading zed sdk was slow and always failed.

willzoe, 2020-05-12:
I also encountered the same problem when installing the SDK `ZED_SDK_Tegra_JP42_v3.1.2.run`.

dibet, 2020-06-12:
Hi, i can install but i have this error when run the example 

[ZED][Init] Depth mode: PERFORMANCE
[ZED][Init] Video mode: HD720@60
[Sample] Object Detection: Loading Module...
[Sample] [Error] enableObjectDetection | CORRUPTED SDK INSTALLATION : Corrupted SDK detected. Please reinstall the ZED SDK.
Exit program.

drproy2k, 2020-08-04:
Hi, I have the same issue(Verifying archive integrity...Error in MD5 checksums... ) with downloading below 3 files
ZED SDK for Jetpack 4.43.2.1 (Jetson Nano, NX, TX2, Xavier, CUDA 10.2)
ZED SDK for Jetpack 4.33.2.1 (Jetson Nano, TX2, Xavier, CUDA 10)
ZED SDK for Jetpack 4.2 3.2.1 (Jetson Nano, TX2, Xavier, CUDA 10)

After downloading these 3 files, each one's file size is around 50MB. I think it should be around 280MB. Am I right?
My storage is enough. Maybe it is also CDN problem..
Please help me.

adujardin, 2020-08-04:
@drproy2k The 3.2 installers are lighters than previous ones, the expected size is between ~30Mo (JP44) and 50Mo for jetsons.

To check if it's a CDN problem you can call the direct links, it's fairly straightforward to guess it (see comment above https://github.com/stereolabs/zed-examples/issues/242#issuecomment-622352122)
For instance, without CDN, the JP44 3.2.1 installer link would be (the base URL won't change) : 
https://stereolabs.sfo2.digitaloceanspaces.com/zedsdk/3.2/ZED_SDK_Tegra_JP44_v3.2.1.run

drproy2k, 2020-08-05:
> @drproy2k The 3.2 installers are lighters than previous ones, the expected size is between ~30Mo (JP44) and 50Mo for jetsons.
> 
> To check if it's a CDN problem you can call the direct links, it's fairly straightforward to guess it (see comment above [#242 (comment)](https://github.com/stereolabs/zed-examples/issues/242#issuecomment-622352122))
> For instance, without CDN, the JP44 3.2.1 installer link would be (the base URL won't change) :
> https://stereolabs.sfo2.digitaloceanspaces.com/zedsdk/3.2/ZED_SDK_Tegra_JP44_v3.2.1.run

Now, I've downloaded ZED SDK from StereoLabs web-site successfully.
The problem's gone.
Thanks anyway for your help.

closed 2022-04-27 by github-actions[bot]