stereolabs/zed-sdk #245: OpenGL: Error while compiling shader

JesperChristensen89, 2020-05-11:
When I try to run any of the examples or samples that includes OpenGL, I get the error: 

```
Error: (999): unknown error
[Sample] ERROR: while compiling shader :  
0:1(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.40, 1.00 ES, and 3.00 ES

[Sample] ERROR: while compiling vertex shader 
[Sample] ERROR: while compiling shader :  
0:1(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.40, 1.00 ES, and 3.00 ES

[Sample] ERROR: while compiling fragment shader 
Segmentation fault (core dumped)
```

I am running on a GPU server with currently 4 GPUs, but the Desktop GUI is running on the motherboard, i.e. it does not use the GPU. 
Do I somehow need to specify which GPU to use, and how do I do this?

obraun-sl, 2020-05-11:
With OpenGL, the GPU used for OpenGL is always the one connected to the display.
On your setup, OpenGL is running on the GPU unit of your CPU (For example on Intel, it's probably an intel HD graphic)

JesperChristensen89, 2020-05-11:
Is there any way to change this? I have no option to change the server graphics.

apanagar, 2021-07-17:
I'm having the same problem. My GPU and camera are connected to an Ubuntu machine, and I'm VNC's into that machine over the network. Therefore, as I understand it, my machine is not connected to a display

closed 2022-04-27 by github-actions[bot]