stereolabs/zed-sdk #247: [ZED Explorer] white screen when trying to play an svo file

amaanda, 2020-05-19:
any ideias on why this might happen? On this example I'm using zed 2.8.3 sdk, on a windows 10 machine.

[image]

obraun-sl, 2020-05-20:
It looks like an opengl issue. 
Do you have the same issue with other tools such as ZED Depth Viewer ?
or in LIVE mode ?

amaanda, 2020-05-20:
Yes, when I try opening the Depth Viewer I get the same blank screen.

amaanda, 2020-05-20:
I tried reinstalling the SDK but it still won't work

amaanda, 2020-05-20:
Apparently the opengl dll is in the folder... The weird thing is: when I use the snapshot button at the Explorer, the saved image is perfectly fine.

[image]

closed 2022-04-27 by github-actions[bot]