stereolabs/zed-sdk #249: Distance Measure

malith1992, 2020-06-08:
Is there any possible way to get the real world distance between two points using the ZED2 camera using ZED python API

obraun-sl, 2020-06-09:
Once you have the 2 points (3D points), you can calculate the euclidean distance like this : 

```
import math
distance = math.sqrt( ((p1[0]-p2[0])**2)+((p1[1]-p2[1])**2)+((p1[2]-p2[2])**2) )
```

where p1 and p2 are your 3D points.

malith1992, 2020-06-09:
> Once you have the 2 points (3D points), you can calculate the euclidean distance like this :
> 
> ```
> import math
> distance = math.sqrt( ((p1[0]-p2[0])**2)+((p1[1]-p2[1])**2)+((p1[2]-p2[2])**2) )
> ```
> 
> where p1 and p2 are your 3D points.

Is the 3D point mean the out put of this function `point_cloud.get_value(x, y)` right ?

obraun-sl, 2020-06-09:
Yes. You can get P1 and P2 like this : 
p1 = point_cloud.get_value(x1, y1);
p2 = point_cloud.get_value(x2, y2);

where x1,y1,x2,y2, must respect [0,Resolution]

See here : https://github.com/stereolabs/zed-examples/blob/master/tutorials/tutorial%203%20-%20depth%20sensing/python/depth_sensing.py#L72

closed 2022-04-27 by github-actions[bot]