stereolabs/zed-sdk #251: CORRUPTED SDK INSTALLATION

dibet, 2020-06-11:
Hi, i'am trying to run the example object detection and the result is the following error:
[ZED][Init] Depth mode: PERFORMANCE
[ZED][Init] Video mode: HD720@60
[Sample] Object Detection: Loading Module... 
[Sample] [Error] enableObjectDetection  | CORRUPTED SDK INSTALLATION : Corrupted SDK detected. Please reinstall the ZED SDK. 
Exit program.

I check previous post and i have one file into the folder resources.
I attach my diagnostic result too.

Thanks.
Dibet
[ZED_Diagnostic_Results.txt](https://github.com/stereolabs/zed-examples/files/4764279/ZED_Diagnostic_Results.txt)

dibet, 2020-06-11:
The installed SDK is ZED_SDK_Tegra_JP43_v3.1.2

obraun-sl, 2020-06-15:
Did you check that the jetpack version was correct? if you use JP43 installer, it needs to be Jetpack 4.3 on the jetson. 
can you post a  $ ldd /usr/local/zed/lib/lisl_ai.so output ?

dibet, 2020-06-15:
/usr/local/zed/lib$ ldd libsl_ai.so 
        linux-vdso.so.1 (0x0000007f8b267000)
        libpthread.so.0 => /lib/aarch64-linux-gnu/libpthread.so.0 (0x0000007f8aba9000)
        libdl.so.2 => /lib/aarch64-linux-gnu/libdl.so.2 (0x0000007f8ab94000)
        librt.so.1 => /lib/aarch64-linux-gnu/librt.so.1 (0x0000007f8ab7d000)
        libpng16.so.16 => /usr/lib/aarch64-linux-gnu/libpng16.so.16 (0x0000007f8ab42000)
        libz.so.1 => /lib/aarch64-linux-gnu/libz.so.1 (0x0000007f8ab15000)
        libjpeg.so.8 => /usr/lib/aarch64-linux-gnu/libjpeg.so.8 (0x0000007f8aacb000)
        libusb-1.0.so.0 => /lib/aarch64-linux-gnu/libusb-1.0.so.0 (0x0000007f8aaa5000)
        libcuda.so.1 => /usr/lib/aarch64-linux-gnu/tegra/libcuda.so.1 (0x0000007f89b80000)
        libnvinfer.so.6 => not found
        libnvinfer_plugin.so.6 => not found
        libstdc++.so.6 => /usr/lib/aarch64-linux-gnu/libstdc++.so.6 (0x0000007f899ec000)
        libm.so.6 => /lib/aarch64-linux-gnu/libm.so.6 (0x0000007f89932000)
        libgomp.so.1 => /usr/lib/aarch64-linux-gnu/libgomp.so.1 (0x0000007f898f5000)
        libgcc_s.so.1 => /lib/aarch64-linux-gnu/libgcc_s.so.1 (0x0000007f898d1000)
        libc.so.6 => /lib/aarch64-linux-gnu/libc.so.6 (0x0000007f89778000)
        /lib/ld-linux-aarch64.so.1 (0x0000007f8b23c000)
        libudev.so.1 => /lib/aarch64-linux-gnu/libudev.so.1 (0x0000007f8974e000)
        libnvrm_gpu.so => /usr/lib/aarch64-linux-gnu/tegra/libnvrm_gpu.so (0x0000007f8970b000)
        libnvrm.so => /usr/lib/aarch64-linux-gnu/tegra/libnvrm.so (0x0000007f896c9000)
        libnvrm_graphics.so => /usr/lib/aarch64-linux-gnu/tegra/libnvrm_graphics.so (0x0000007f896aa000)
        libnvidia-fatbinaryloader.so.32.2.0 => /usr/lib/aarch64-linux-gnu/tegra/libnvidia-fatbinaryloader.so.32.2.0 (0x0000007f8964c000)
        libnvos.so => /usr/lib/aarch64-linux-gnu/tegra/libnvos.so (0x0000007f8962e000)

dibet, 2020-06-15:
Iam using ZED_SDK_Tegra_JP43_v3.1.2.run

dibet, 2020-06-15:
Hi, now iam installing ZED_SDK_Tegra_JP43_v3.2.0.run and i have this error:
Detected Tegra_JP42, required exact Tegra_JP43 .
What is the meaning of that?

closed 2020-06-20 by dibet