stereolabs/zed-sdk #254: One doubt in camera stream.

dibet, 2020-06-14:
Hi after some time transmitting over network, i catch the following error : CAMERA_NOT_INITIALIZED.
Any help?

obraun-sl, 2020-06-15:
Hi,
This error is triggered when no buffer are received through the network for a certain amount of time. (more than 3/4  seconds). 
Let me know if you have the same behavior in local streaming (same PC to same PC), to check if it can comes from something else.
We will increase the timeout for receiving in the next version since it can be low on specific network.

dibet, 2020-06-20:
Olá i'am testing the network streaming with 
    init_parameters.camera_resolution = sl::RESOLUTION::HD720;
    init_parameters.depth_mode = DEPTH_MODE::PERFORMANCE;
    init_parameters.camera_fps = 60;

The streaming is running at 45 fps in a Jetson Nano.

When i call the following lines in the receiver:

    double e1 = cv::getTickCount();
    auto returned_state = zed.grab();
    double e2 = cv::getTickCount();
   LOG(INFO) << "Delay " << (e2 - e1) / cv::getTickFrequency();

I have the following output

I0620 11:59:33.022883 14016 cvcamerastereolabs.cpp:111] Delay 0.205931
I0620 11:59:33.094099 14016 cvcamerastereolabs.cpp:111] Delay 0.00156142
I0620 11:59:33.131688 14016 cvcamerastereolabs.cpp:111] Delay 0.00151818
I0620 11:59:33.447721 14016 cvcamerastereolabs.cpp:111] Delay 0.279961
I0620 11:59:33.518149 14016 cvcamerastereolabs.cpp:111] Delay 0.0015011
I0620 11:59:33.731513 14016 cvcamerastereolabs.cpp:111] Delay 0.176742
I0620 11:59:33.787251 14016 cvcamerastereolabs.cpp:111] Delay 0.00172908
I0620 11:59:33.827555 14016 cvcamerastereolabs.cpp:111] Delay 0.00160762
I0620 11:59:34.056102 14016 cvcamerastereolabs.cpp:111] Delay 0.191044
I0620 11:59:34.125304 14016 cvcamerastereolabs.cpp:111] Delay 0.00151111

Then my fps are reduced 5.45704 fps in the receiver
Do you know the cause of this problem. I have good netkwork connection.

dibet, 2020-06-20:
In other issue i have problem with the object_detection module, then i need to install JP42 and it work, now i install JP43 v 3.1.2 and i have 25 fps but never 60fps.

dibet, 2020-07-06:
Another doubt. If can i change the stream parameters to
init_parameters.camera_fps = 45; 
for example. 
I am trying to eliminate the error CAMERA_NOT_INITIALIZED.

obraun-sl, 2020-07-07:
You can't. Available FPS are 15;30;60 or 100 (VGA only).
In the StreamingParameters structure, you can set a streaming FPS (if you want it different from camera FPS). streaming fps and camera fps must be multiple then.

closed 2022-04-27 by github-actions[bot]