stereolabs/zed-sdk #257: About Streaming, Why is this timeout happen?

muraiyuya, 2020-06-22:
-------------------------------------------------------------------------------------------------------
Opening the stream requires the IP of the sender
Usage : ./ZED_Streaming_Receiver IP:[port]
You can specify it now, then press ENTER, 'IP:[port]': 30000
[ZED][Streaming] Error: Stream opening timeout
[Sample][Error] Camera Open  | FAILURE : The function does not perform as expected. Exit program.
[ZED]ERROR: sl::Camera::Open has not been called, no Camera instance running.
----------------------------------------------------------------------------------------------------------
I use Jetson xavier NX with zed mini as a streaming Sender, and  MSI laptop as a Receiver.
I checked anything other functions of SDK and played well, but just only streaming function is not working now.
I see this comand in Receiver screen, and I don't know the solution for it. 
Please help!!!

obraun-sl, 2020-06-22:
You need to enter the IP , the port is optional. Here you just set the port (30000)
If you know the IP of the NX, then your receiving command should be : 
`./ZED_Streaming_Receiver <IP>:<port>`
For example, 
`./ZED_Streaming_Receiver 192.168.1.31:30000
`

muraiyuya, 2020-06-22:
I see!
Thank you for fast reply!
How about sender command?
./ZED_Streaming_Receiver <IP>:<port>
For instance
./ZED_Streaming_Sender 192.168.??.??:30000
like this?

muraiyuya, 2020-06-22:
I see!
Thank you for fast reply!
How about sender command?
./ZED_Streaming_Sender <IP>:<port> :
For instance
./ZED_Streaming_Sender 192.168.??.??:30000
like this?

muraiyuya, 2020-06-22:
I see!
Thank you for fast reply!
How about sender command?
./ZED_Streaming_Sender <IP><port>: 
For instance
./ZED_Streaming_Sender 192.168.??.??:30000
like this?

obraun-sl, 2020-06-22:
the sender does not need to know the IP. You can specify the port if you want but it must be the same as receiver.

muraiyuya, 2020-06-22:
Thank you so much .
I'm trying now.

muraiyuya, 2020-06-22:
And it works!
Thanks a lot.

closed 2020-06-22 by muraiyuya