stereolabs/zed-sdk #258: Can I get a spatial mesh map by streaming ?

muraiyuya, 2020-06-22:
We can see camera live video by streaming by SDK initial option, but can we get spatial mesh map by streaming ? 
How can I make it real?
I want to use ZED mini on drone , and watch spatial mesh map sent by flying drone in real time.

obraun-sl, 2020-06-23:
Use the Spatial mapping sample and change the input to : 
`init_parameters.input.setFromStream(ip,port);`

muraiyuya, 2020-06-23:
Okay.
So,how should I change the code and where to chang?
I'm pasting the "Spatial Mapping code" below.
-----------------------------------------------------------------------------------------------------
 if (sscanf(arg.c_str(),"%u.%u.%u.%u:%d", &a, &b, &c, &d,&port) == 5) {
            // Stream input mode - IP + port
            string ip_adress = to_string(a)+"."+to_string(b)+"."+to_string(c)+"."+to_string(d);
            param.input.setFromStream(sl::String(ip_adress.c_str()),port);
            cout<<"[Sample] Using Stream input, IP : "<<ip_adress<<", port : "<<port<<endl;
        }
        else  if (sscanf(arg.c_str(),"%u.%u.%u.%u", &a, &b, &c, &d) == 4) {
            // Stream input mode - IP only
            param.input.setFromStream(sl::String(argv[1]));
            cout<<"[Sample] Using Stream input, IP : "<<argv[1]<<endl;
        }
----------------------------------------------------------------------------------------------------------

muraiyuya, 2020-06-23:
Okay.
So,how should I change the code and where to chang?
I'm pasting the "Spatial Mapping code" below.

-------------
if (sscanf(arg.c_str(),"%u.%u.%u.%u:%d", &a, &b, &c, &d,&port) == 5) {
// Stream input mode - IP + port
string ip_adress = to_string(a)+"."+to_string(b)+"."+to_string(c)+"."+to_string(d);
param.input.setFromStream(sl::String(ip_adress.c_str()),port);
cout<<"[Sample] Using Stream input, IP : "<<ip_adress<<", port : "<<port<<endl;
}
else if (sscanf(arg.c_str(),"%u.%u.%u.%u", &a, &b, &c, &d) == 4) {
// Stream input mode - IP only
param.input.setFromStream(sl::String(argv[1]));
cout<<"[Sample] Using Stream input, IP : "<<argv[1]<<endl;
}

obraun-sl, 2020-06-23:
Since we already provide the way to use streaming input in the spatial mapping sample, you don't have to change anything. Just launch this way : 
$ ./ZED_Spatial_Mapping <ip>:port

For example, 
$ ./ZED_Spatial_Mapping 192.168.1.60:30000

muraiyuya, 2020-06-23:
I see!
Thank you so much.

$ ./ZED_Spatial_Mapping 192.168.1.60:30000
Above comand should be at Receiver side?
How about Sender side?

obraun-sl, 2020-06-23:
Receiver : 
$ ./ZED_Spatial_Mapping 192.168.1.60:30000

Sender  (no need to specify IP)
$ ./ZED_Streaming_Sender

muraiyuya, 2020-06-23:
I see.
So at the Receiver side, which should I tipe "$ ./ZED_Spatial_Mapping 192.168....(port)" in the directory,
\spatial mapping\cpp\build 
or
 \camera streaming\receiver\cpp\build ???

muraiyuya, 2020-06-23:
My receiver side is visual studio 2019, how can I represent the comand code ,$ ./ZED_Spatial_Mapping 192.168.1.60:30000?

obraun-sl, 2020-06-23:
$ is just to indicate that it must run in a terminal...
On windows, the syntax will be more (in a terminal ) : 

`ZED_Spatial_Mapping.exe 192.168.1.60:30000`

Zgihu, 2020-06-24:
I understand it now.
Thanks a lot!

closed 2022-04-27 by github-actions[bot]