stereolabs/zed-sdk #261: Unintentional stderr output in zed-config.cmake

spiderkeys, 2020-06-27:
SDK version: 3.2.0, CUDA 10.2, Ubuntu 18.04

Using cmake's message() without the STATUS mode results in messages being sent to stderr. See:
https://cmake.org/cmake/help/v3.0/command/message.html 

There is a bit of code in zed-config.cmake that uses message in this way, when it should probably use STATUS:

```cmake
message("-- Found TensorRT (found version "${TENSORRT_LIBRARY_VERSION_MAJOR}.${TENSORRT_LIBRARY_VERSION_MINOR}\")")
```

This pollutes my build output which looks for packages that write to stderr. Easily patched on my end, but should be corrected, since this just seems to be a debug/informational message of sorts.

spiderkeys, 2020-06-27:
There appear to be two additional instances of this:
```cmake
 message("${TENSORRT_STATIC_DEP_LIBRARIES}")
 message("${TENSORRT_STATIC_DEP_LIBRARIES}")
```

adujardin, 2020-06-29:
Thanks for reporting the issue, you're right.
It has been fixed and will be available in the next release.

closed 2020-06-29 by spiderkeys