stereolabs/zed-sdk #263: Unable to Retrieve Temperature Data

birdistheword96, 2020-07-01:
I have a ZED2 Camera (Camera Firmware: 1523, Sensors Firmware: 1523) connected to a Nvidea Jetson Xavier running Jetpack 4.3 (Ubuntu 18.04.4 LTS with python 3.6 installed) and the latest ZED SDK (3.2).

I am currently unable to retrieve any Temperature measurements from any of the onboard temperature sensors, using either C++ or python. 

When I use the following code in C++, they all return a 'NAN' value:
```c++
sl::SensorsData::TemperatureData temperature_data;
float temperature_left, temperature_right, temperature_imu, temperature_barometer;

temperature_data.get(sl::SensorsData::TemperatureData::SENSOR_LOCATION::ONBOARD_LEFT, temperature_left);
		temperature_data.get(sl::SensorsData::TemperatureData::SENSOR_LOCATION::ONBOARD_RIGHT, temperature_right);
		temperature_data.get(sl::SensorsData::TemperatureData::SENSOR_LOCATION::IMU, temperature_imu);
		temperature_data.get(sl::SensorsData::TemperatureData::SENSOR_LOCATION::BAROMETER, temperature_barometer);
		printf("Left: %.2f, Right: %.2f, IMU: %.2f, Barometer: %.2f\r\n", temperature_left, temperature_right,
																	  temperature_imu, temperature_barometer);
```

And the Equivalent in python just returns -1 for all values:
```python
temperature_data = sl.TemperatureData()

temperature_left = temperature_data.get(sl.SENSOR_LOCATION.ONBOARD_LEFT)
temperature_right = temperature_data.get(sl.SENSOR_LOCATION.ONBOARD_RIGHT)
temperature_imu = temperature_data.get(sl.SENSOR_LOCATION.IMU)
temperature_barometer = temperature_data.get(sl.SENSOR_LOCATION.BAROMETER)

print("Left: {:.2f}, Right: {:.2f}, IMU: {:.2f}, Barometer: {:.2f}\r\n".format(temperature_left, temperature_right,
                        temperature_imu, temperature_barometer))
```

For reference I can access all the values for the IMU, Barometer and Magnetometer, it's just the temperature values I can't get (ironically the only ones I need)

What am I doing wrong? And how can I access these values?

obraun-sl, 2020-07-07:
Hi,
I did not manage to reproduce it. 
How do you call getSensorsData() (which parameters)
Also, what is the sensors firmware (1523 is the camera fw, it should be 776 or 777) ?

birdistheword96, 2020-07-07:
The version returned when I call the camera information function is the same firmware number:

```python
>>> info = zed.get_camera_information()
>>> print("Camera Model: " + str(info.camera_model))
Camera Model: ZED 2
>>> print("Camera Firmware: " + str(info.camera_configuration.firmware_version))
Camera Firmware: 1523
>>> print("Sensors Firmware: " + str(info.sensors_configuration.firmware_version))
Sensors Firmware: 1523
```

For the getSensorsData(), I do the following in python:
```python
sensors_data = sl.SensorsData()
temperature_data = sl.TemperatureData()
if zed.get_sensors_data(sensors_data, sl.TIME_REFERENCE.CURRENT) == sl.ERROR_CODE.SUCCESS :
    linear_acceleration = sensors_data.get_imu_data().get_linear_acceleration()
    temperature_left = temperature_data.get(sl.SENSOR_LOCATION.ONBOARD_LEFT)
    temperature_right = temperature_data.get(sl.SENSOR_LOCATION.ONBOARD_RIGHT)
    temperature_imu = temperature_data.get(sl.SENSOR_LOCATION.IMU)
    temperature_barometer = temperature_data.get(sl.SENSOR_LOCATION.BAROMETER)
    print("Left: {:.2f}, Right: {:.2f}, IMU: {:.2f}, Barometer: {:.2f}\r\n".format(temperature_left, temperature_right,
												  temperature_imu, temperature_barometer))
```

and in C++
```c++
sl::InitParameters init_params;
init_params.coordinate_units = sl::UNIT::METER;
init_params.sensors_required = true;

sl::SensorsData::TemperatureData temperature_data;
temperature_data = sensor_data.temperature;
float temperature_left, temperature_right, temperature_imu, temperature_barometer;

temperature_data.get(sl::SensorsData::TemperatureData::SENSOR_LOCATION::ONBOARD_LEFT, temperature_left);
temperature_data.get(sl::SensorsData::TemperatureData::SENSOR_LOCATION::ONBOARD_RIGHT, temperature_right);
temperature_data.get(sl::SensorsData::TemperatureData::SENSOR_LOCATION::IMU, temperature_imu);
temperature_data.get(sl::SensorsData::TemperatureData::SENSOR_LOCATION::BAROMETER, temperature_barometer);
printf("Left: %.2f, Right: %.2f, IMU: %.2f, Barometer: %.2f\r\n", temperature_left, temperature_right,
											temperature_imu, temperature_barometer);
```

But I just get NAN values for the temperature values.

I Must be doing something wrong but I can't work out what it is!

Thanks in advance!

yy2lyx, 2021-08-16:
@birdistheword96 I also meet the same issue! Do U solve it ?

yy2lyx, 2021-08-16:
@birdistheword96 oh，I work it out! Here is the code!
`sensors_data = sl.SensorsData()
if zed.get_sensors_data(sensors_data, sl.TIME_REFERENCE.CURRENT) == sl.ERROR_CODE.SUCCESS:
    	temperature_left = sensors_data.get_temperature_data().get(sl.SENSOR_LOCATION.ONBOARD_LEFT)
        temperature_right = sensors_data.get_temperature_data().get(sl.SENSOR_LOCATION.ONBOARD_RIGHT)
        temperature_imu = sensors_data.get_temperature_data().get(sl.SENSOR_LOCATION.IMU)
        temperature_barometer = sensors_data.get_temperature_data().get(sl.SENSOR_LOCATION.BAROMETER)
        print("Left: {:.2f}, Right: {:.2f}, IMU: {:.2f}, Barometer: {:.2f}\r\n".format(temperature_left,
                                                                                       temperature_right,
                                                                                       temperature_imu,
                                                                                  temperature_barometer))`

Myzhar, 2021-11-03:
Closed for inactivity

closed 2021-11-03 by Myzhar