stereolabs/zed-sdk #266: version of OpenNI2 dependency

templarares, 2020-07-16:
Hi Stereolabs,

I am using 'camera streaming' Sender on a Jetson TX2 with ZED SDK 3.1 to stream ZED data to a remote host. That works well. Yet I am also planning to deploy another application (using another Cam) on the same TX2 and it turned out that its version of OpenNI2 dependency conflicts with that of ZED SDK 3.1 and the two applications cannot be compiled at the same time. I used to work with ZED SDK 2.8 and there was no such conflict, but SDK 2.8 is extremely flawed in streaming so I cannot go back to that. I was wondering if there is any other alternative to solve this conflict? thank!

obraun-sl, 2020-07-16:
Can you precise the conflict you have (error log) when building your app with OpenNI and the app with 3.1 ?

Myzhar, 2021-11-03:
Closed for inactivity

closed 2021-11-03 by Myzhar