stereolabs/zed-sdk #269: Body Tracking: Visualization of different poses

aozcan97, 2020-08-05:
I'm trying to find a way with the source code for C++ how to evaluate the position of the arms and legs and output it in the visualization window (GLViewer). My goal is to detect poses like the arms are up or down. @nesnes @adujardin @obraun-sl @P-yver

obraun-sl, 2020-08-05:
I think that what you want is done in the sample Body Tracking : 
https://github.com/stereolabs/zed-examples/tree/master/body%20tracking/cpp
It shows the bones in the OpenGL window (in 3D)

aozcan97, 2020-08-05:
You are right the skeleton is displayed correctly, but I still want recognition and an output that the arms are in the air or not. As well as the ID and the speed which is above the head of the person.

aozcan97, 2020-08-10:
I found these links on the Stereolabs website, but they don't really show anything about the position of the joints or the position of the links of the body parts. @obraun-sl 
https://www.stereolabs.com/docs/api/group__Object__group.html#gacce1373b1af32c33563d484bf326bc8e

https://www.stereolabs.com/docs/api/group__Object__group.html#ga87fd98be8a135ca065216e49ff6e4fdf

obraun-sl, 2020-08-11:
Hi. Here is a picture for the Body18 keypoints 
[image]

closed 2022-04-22 by aozcan97