stereolabs/zed-sdk #270: i can not get the disparity map,it confused me.

kearcer, 2020-08-08:
dear author! I want to use the following code to get the disparity map 

cap.retrieve_measure(disparity, sl.MEASURE.DISPARITY, resolution=res)
cv2.imwrite('save/disp/depth' + str(a) + '.png', disparity.get_data())

but it fails.The output image is all black. if i change the "DISPARITY" to "DEPTH" it can output the depth map correctly, but the disparity map does not work. Does anyone know what went wrong? 
PS: I output the matrix of the acquired image, many pixels are negative numbers.
If someone can solve the problem,i am very appreciate!!!

obraun-sl, 2020-08-09:
Hi,
Disparity values are always negative in stereoscopic parallel system (which is the case or the ZED). 
That is why it's black because values will be clamped to 0. 
If you want to display a disparity map, you need to multiply by -1 and convert to uchar format.

kearcer, 2020-08-11:
@obraun-sl Thank you for your reply, but I would like to know which way to get the depth after obtaining the disparity map. I don’t want to use the camera's own method to directly generate the depth map. I used the following method
depth = cv2.reprojectImageTo3D(disparity.astype(np.float32)*256, Q)
 which can make me get approximately correct The results such as this:
[image]
but there are still some problem with the camera get the distance.
[image]
if someone can solve the problem,i am very appreciate for you ！！！！

obraun-sl, 2020-08-17:
reprojectImageTo3D will give proper results if the Q matrix is correct. I don't know how you get the Q matrix as it should be extracted from stereoRectify from opencv. 
The relation between disparity and depth is simpler in our case : 

depth = focal * baseline / disparity, where 
```
focal = zed.getCameraInformation().camera_configuration.calibration_parameters.left_cam.fx;
baseline = zed.getCameraInformation().camera_configuration.calibration_parameters.baseline;
```

Wei10857104, 2020-08-19:
I also have this kind of problem and I'm still paying attention

obraun-sl, 2020-08-19:
Does the provided answer help ? otherwise, please open another issue

closed 2022-04-27 by github-actions[bot]