stereolabs/zed-sdk #275: how to generate disparity map on python or c++?

Wei10857104, 2020-08-19:
I want to get result like this. 

[image]

But i use python depth viewer's sample code .

Although the environment is completely different, I at least hope to get similar results.

[image]

This is my python's code.
`import pyzed.sl as sl
import math
import numpy as np
import sys
import cv2
import matplotlib.pyplot as plt

# Create a Camera object
zed = sl.Camera()

# Create a InitParameters object and set configuration parameters
init_params = sl.InitParameters()
init_params.depth_mode = sl.DEPTH_MODE.PERFORMANCE  # Use PERFORMANCE depth mode
init_params.coordinate_units = sl.UNIT.METER  # Use meter units (for depth measurements)
init_params.camera_resolution = sl.RESOLUTION.HD720

# Open the camera
err = zed.open(init_params)
if err != sl.ERROR_CODE.SUCCESS:
    exit(1)

# Create and set RuntimeParameters after opening the camera
runtime_parameters = sl.RuntimeParameters()
runtime_parameters.sensing_mode = sl.SENSING_MODE.STANDARD  # Use STANDARD sensing mode
# Setting the depth confidence parameters
runtime_parameters.confidence_threshold = 100
runtime_parameters.textureness_confidence_threshold = 100

# Capture 150 images and depth, then stop
i = 0
image = sl.Mat()
depth = sl.Mat()
point_cloud = sl.Mat()

mirror_ref = sl.Transform()
mirror_ref.set_translation(sl.Translation(2.75,4.0,0))
tr_np = mirror_ref.m

while(1):
    if zed.grab(runtime_parameters) == sl.ERROR_CODE.SUCCESS:
         # Retrieve left image
        zed.retrieve_image(image, sl.VIEW.LEFT)
        image_ocv = image.get_data()
        # Retrieve depth map. Depth is aligned on the left image
        zed.retrieve_measure(depth, sl.MEASURE.DEPTH)
        # Use get_data() to get the numpy array
        image_ocv = depth.get_data()
        # Display the left image from the numpy array
        cv2.imshow("Image", image_ocv)
        if cv2.waitKey(1) & 0xFF == ord('q'):
            break

        # Retrieve colored point cloud. Point cloud is aligned on the left image.
        zed.retrieve_measure(point_cloud, sl.MEASURE.XYZRGBA)
        # Use get_data() to get the numpy array
        point_cloud_ocv = point_cloud.get_data()
        # Display the left image from the numpy array
#         cv2.imshow("Image", point_cloud_ocv)
#         if cv2.waitKey(1) & 0xFF == ord('q'):
#             break        

# Close the camera

cv2.destroyAllWindows()
zed.close()`

Wei10857104, 2020-08-19:
Sorry for typographic issues, I am a newbie for that.

obraun-sl, 2020-08-19:
You want to display a depth image. Therefore it's better to use : 
`zed.retrieve_image(depth, sl.VIEW.DEPTH)`
instead of 
`zed.retrieve_measure(depth, sl.MEASURE.DEPTH)` (since this is not an image but a float matrix)

Wei10857104, 2020-08-21:
Thanks. It helps me solve problem.

closed 2020-08-21 by Wei10857104