stereolabs/zed-sdk #276: Stream remotely over 5G network

BingXiong1995, 2020-08-19:
Our team is using ZED2 on a drone to do real-time remote spatial mapping. The drone will fly far away so a local network streaming might not be available. We used a large bandwidth server(100Mbs) with a public IP and using a 5G network card on our drone to establish a remote network transmission. We have tried a lot of method such as port forwarding and VPN to get the streaming from the drone over the network, but all of them are failed. Is there any suggestion to get stream over a server with public IP ?
@obraun-sl  Thanks for your reply!

obraun-sl, 2020-08-20:
If you are using the streaming module from the ZED SDK, you need to make sure that the sender and receiver are on the same network. 
Otherwise, you might be interested in our new gstreamer plugin here : 
https://github.com/stereolabs/zed-gstreamer

BingXiong1995, 2020-08-20:
@obraun-sl We'll try that later. Thanks for your amazing stereo camera!

Vertigo-VR, 2021-03-04:
Hello.

Any update to using ZED for remote spatial mapping? Our use case is much alike yours @BingXiong1995, did you come up with any good solutions? Thanks.

keli95566, 2022-01-06:
Hi @BingXiong1995 , I am wondering if you have tried the Gstreamer option and is able to retrieve the images from Unity correctly? Thank you!

closed 2022-04-27 by github-actions[bot]