stereolabs/zed-sdk #283: some problems about object detection

Zhang-Qinghui, 2020-09-10:
Hello，why do I keep getting stuck at the step of downloading the model when using the 3d detection example.As shown in the picture
[image]

obraun-sl, 2020-09-10:
Hi,
The model to be download is around 200Mb heavy. Make sure you are waiting enough according to your internet speed.
Once downloaded, the module will start to optimize the model for you graphic card.

Zhang-Qinghui, 2020-09-10:
Thanks for your reply. You are right. But when I downloaded the model and ran it again, there was another error“段错误（核心已转储）”. Why does this happen？
[image]

obraun-sl, 2020-09-10:
Can you give more precision about the graphic card used ?

Zhang-Qinghui, 2020-09-11:
Of course.The graphic card I used is GTX 750Ti. Is it because of the graphics card?

obraun-sl, 2020-09-11:
It might come from that. The GTX750 does not have a lot of GPU memory and AI model does require a lot. 
You can try to reduce the GPU memory used by the depth map by setting the depth mode to PERFORMANCE instead of ULTRA first. 
Then the HUMAN_BODY_FAST model has less requirement on GPU memory and might be a better fit for your graphic card.

Zhang-Qinghui, 2020-09-11:
Thanks for your reply. Now I try to use tx2 to run 3d detection example. Could you please tell me where the model will be downloaded?

closed 2022-04-27 by github-actions[bot]