stereolabs/zed-sdk #284: Why can't I run the 3d object detection example？

Zhang-Qinghui, 2020-09-12:
I use RTX 2060 GPU to run 3d object detection example. But this problem occurred “段错误（核心已转储）”. I do not why this happen. Could you please help me to solve this? Thank you in advance.
[image]

phdsky, 2020-09-22:
➜  build ./ZED_Object_Detection 
[Sample] Object Detection: Loading Module... 

[ZED][Object Detection] AI model not found, downloading   Done[1]    121608 segmentation fault (core dumped)  ./ZED_Object_Detection

Same to @Zhang-Qinghui , I also encountered this problem, crashed in SDK retrieveObjects function.  

On ubuntu 20.04, CUDA 11.0, Quadro RTX 4000 8G, ZED 2

@obraun-sl  Any update? Thanks

obraun-sl, 2020-09-22:
Hi,
It seems that the optimization of the AI model crash. 
Try again from scratch and make sure that you have enough GPU memory available before starting. 

1- Remove the temp files : 
```
$ sudo rm -R /tmp/.x6G* 
$ sudo rm -rf /usr/local/zed/resources/*
```

2- Start the object detection sample again. 
It should re-download and retry the optimization.

phdsky, 2020-09-22:
> Hi,
> It seems that the optimization of the AI model crash.
> Try again from scratch and make sure that you have enough GPU memory available before starting.
>
> 1- Remove the temp files :
>
> ```
> $ sudo rm [...]

Just tried, but not work. Also crashed in SDK **retrieveObjects** function.

I'm not sure how much GPU memory is enough? 

7G+ is not enough yet? Thanks for u reply

Tue Sep 22 15:30:44 2020       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 450.66       Driver Version: 450.66       CUDA Version: 11.0     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  Quadro RTX 4000     Off  | 00000000:65:00.0  On |                  N/A |
| 30%   35C    P8    10W / 125W |    353MiB /  7973MiB |      3%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A      1059      G   /usr/lib/xorg/Xorg                 39MiB |
|    0   N/A  N/A      1906      G   /usr/lib/xorg/Xorg                108MiB |
|    0   N/A  N/A      2153      G   /usr/bin/gnome-shell              151MiB |
|    0   N/A  N/A    128335      G   ...clion-2020.1/jbr/bin/java       44MiB |
+-----------------------------------------------------------------------------+

lyudongliang, 2020-12-11:
I use ZED2 on nvidia jetsion nx platform using the python API (zed-examples/object detection/image viewer/python/object_detection_image_viewer.py). Then abort or appear segment fault. The SDK is ZED_SDK_Tegra_JP44_v3.3.3.run. 

Opening in BLOCKING MODE 
NvMMLiteOpen : Block : BlockType = 279 
NVMEDIA: Reading vendor.tegra.display-size : status: 6 
NvMMLiteBlockCreate : Block : BlockType = 279 
libv4l2_nvvidconv (0):(802) (INFO) : Allocating (9) OUTPUT PLANE BUFFERS Layout=1
libv4l2_nvvidconv (0):(818) (INFO) : Allocating (9) CAPTURE PLANE BUFFERS Layout=0
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
Aborted (core dumped)

closed 2022-04-27 by github-actions[bot]