stereolabs/zed-sdk #285: [ZED]INFO: Sensors not detected

sAgmin, 2020-09-14:
[image]

What does '[ZED]INFO: Sensors not detected' mean?
Is the inner sensor such as IMU etc. broken?
I'm using ZED 2 stereo camera..

Please tell me the solution what can deal with it
Object detection API doesn't work...

obraun-sl, 2020-09-15:
Can you type lsusb and check that it has 2b03:F781 as output (that's the sensor MCU) ?

sAgmin, 2020-09-29:
> Can you type lsusb and check that it has 2b03:F781 as output (that's the sensor MCU) ?

It is fixed automatically..... I don't know why
I suspect that the camera has been used for a long time (3hr? or more?).

closed 2022-04-27 by github-actions[bot]