stereolabs/zed-sdk #286: Why not available on ZED mini???

Tetsujinfr, 2020-09-18:
Hi

Why the object detection functionality in the sdk is limited to the ZED2? I do not understand what is not good enough with the ZED Mini to use it too.

I am very interested by this feature but given I bought a ZED mini recently (due to the small form factor which is a must for me) I feel I am left behind on this.

Can you not enable this feature on the Mini as well?
Thanks

obraun-sl, 2020-09-21:
Hi,
The main reasons are the following : 
- dataset used to train the depth learning model has been optimized with ZED2 images which has a larger FoV than ZED and ZED-M. It gives much better results then on ZED2 than ZED or ZED-M. 
- The IMU is used in combination with deep learning for object detection (for some object) to fix gravity in object tracking. ZED does not have integrated IMU.
- Highest depth range is required for the 3D application, and the ZED2 provides a better range than the ZED-M. Especially for HUMAN_BODY models available since 3.2 (skeleton tracking).
For the moment, there is no plan to optimize models/code for ZED-M as well. 

Note that for ZED and ZED-M, you can use zed-yolo repository that provides integration of ZED/ZED-M/ZED2 with YOLO standard models.
https://github.com/stereolabs/zed-yolo

Tetsujinfr, 2020-09-21:
For the IMU I know, but I have a ZED mini and I bought it over a ZED2 because of 1) its form factor and 2) it has the IMU as well for the optimized tracking or other sensor fusion application, so cool enough piece of hardware for me.
Most of the value on the ZED Mini camera to me is with the algos and api which comes with it, but right now it feels its a two class citizen world (ZED2 vs the rest).
For the training I understand the current costraints but you could train on ZED Mini images as well. I appreciate that is probably quite some work but why not supporting the ZED mini as well as the ZED2 w.r.t deep learning functionalities?
Regarding Yolo, yes I am looking at it but:
1) why would I have to end up doing that work (probably not as well as you guys) while I would expect to benefit from the deeplearning functionalities you develop anyway. Again, the key value proposal on your cameras, for me, is not so much with the hardware but with the software stack you provide.
2) the ZED support for Yolo is quite opaque (some util lib is used where I do not find source code for), no documentation, slow like hell, no object ID instatiation, no trajectory tracking : so basically a ton of work required.

Pls let me know if you intend to either provide deeplearning functionalities for the ZED Mini in future SDK releases, or if you can point me to some good documentation on how using the ZED best along with Yolov4.

Thanks for your help on this, I want to use my Mini with deeplearning algos.

Tetsujinfr, 2020-10-26:
Any plan to port object detection to Zed Mini?

closed 2022-04-27 by github-actions[bot]