stereolabs/zed-sdk #289: Create Pointcloud of the entire environment

Zumbalamambo, 2020-09-26:
Is there any examples on creating pointcloud from the rgbd images that I have?... I'm able to create pointcloud form single rgb and a depth image but how can I retrieve the pointcloud of the complete environment using the rgbd images that I have captured using zed?

obraun-sl, 2020-09-28:
For that, you can use the spatial mapping example with fused point cloud option : 
https://github.com/stereolabs/zed-examples/blob/master/spatial%20mapping/cpp/src/main.cpp#L39

Set the define to 0.

Zumbalamambo, 2020-09-28:
@obraun-sl isn''t it possible to do the same with Python?

Zumbalamambo, 2020-09-28:
executing the `main.cpp` throws the following error,

```
main.cpp:29:10: fatal error: sl/Camera.hpp: No such file or directory
   29 | #include <sl/Camera.hpp>
      |          ^~~~~~~~~~~~~~~
compilation terminated.
```

I have installed zed sdk though. Is there any cmakelist file or something to help me compile the code?

Zumbalamambo, 2020-09-28:
@obraun-sl trying to build the tutorial throws the following error,,

```
CMake Error at /usr/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:146 (message):
  Could NOT find CUDA: Found unsuitable version "10.1", but required is at
  least "11.0" (found /usr)
Call Stack (most recent call first):
  /usr/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:391 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.16/Modules/FindCUDA.cmake:1104 (find_package_handle_standard_args)
  /usr/local/zed/zed-config.cmake:72 (find_package)
  CMakeLists.txt:21 (find_package)
```

I have manually mapped the cuda path but still it is not picking up

[image]

closed 2021-02-21 by Zumbalamambo