stereolabs/zed-sdk #300: Access the ZED camera from multiple apps

Tetsujinfr, 2020-10-26:
Hi

Is it possible to grab the Zed camera frames from multiple apps running in parallel, and in different languages?
E.g. I want to run the SLAM, so I open tje camera and track, but let s say I want to grab frames in a pythin program which rins at the same time as the SLAM: can I do that?

For now, when I open the access to the camera, it seems it locks its usage and I can not access the camera from other programs, unless I stop the one which opened the camera first.

Thanks

adujardin, 2020-10-27:
Technically you can with a trick. 

The idea is to use the streaming feature for the ZED SDK 
- One program is opening the camera and sending the stream of images, 
- All others will open the stream (opening the local network stream).

The opening apps can use the same stream several times and in different languages (using the ZED SDK). Please note that the streams are compressed using hardware acceleration and are therefore limited, but for several apps, it will be fine.

See the streaming samples here https://github.com/stereolabs/zed-examples/tree/master/camera%20streaming

Tetsujinfr, 2020-10-28:
so I gave it a try, and that works however this is quite slow. I am on Jetson Xavier NX, and turning off CV2 image rendering I get get only 15-20fps for 720p@60, with some nasty compression artefacts.
I suspect the h264 encoding, even if helped by the GPU, is the limitation but not sure.

`-> Does it look normal to you? Have you tried the streaming performance on Jetson Nano or NX? Do I do something wrong?`

I need real high fps for my main app (which works fine with your cpp slam example for instance with 100fps refresh in VGA), then I need some reasonable latency like 10 to 20 fps for other apps to grab frames in // of the main app). Now the problem is that the streaming with one server/client pair already put the NX on its knees, without any SLAM or other computer vision workload so there is no point at the stage using it for me.

Thanks a lot for your help

adujardin, 2020-10-28:
The app that sends the stream should do nothing else to avoid issues, especially OpenCV display (which is notoriously slow on jetson), depth computation, etc. 

By default, a ZED SDK instance will compute the depth map in `ULTRA` mode with `depth_stabilization` which enabled the positional tracking to run. None of these should run with the steaming apps (using `DEPTH_MODE::NONE` without any image retrieval, especially CPU) and ideally only one of the app should do it since the computations aren't shared. So 2 instances with depth map are equivalent to 2 independent cameras on the same jetson.

Local streaming to share a single USB ZED is a trick and is not ideal for performance concerns (as using display and python API).

Tetsujinfr, 2020-11-11:
hi, can you please remind me if the zed sdk on jetson would store the zed frames in central ram or in gpu ram , when the gpu cuda acceleration is ON?
trying to see if I can use shared memory as an ipc mechanism here.
thanks

closed 2022-04-27 by github-actions[bot]