stereolabs/zed-sdk #308: depth sensing issues

adetails, 2020-11-16:
issues:AttributeError: type object 'pyzed.sl.Translation' has no attribute 'dot_translation'

P-yver, 2020-11-16:
Hi,
This sample use the latest 3.3.1 release of the Python wrapper.
wrapper: https://github.com/stereolabs/zed-python-api
ZED SDK: https://www.stereolabs.com/developers/release/

madasebrof, 2020-11-17:
I'm confused about the previous reply. 

I just downloaded the latest release (3.3.1) and am having the same problem:

(after running `ZED_SDK_Ubuntu18_full_cuda10.2_v3.3.1.run`)

```
(base) adam@dual-18:/usr/local/zed$ python get_python_api.py
CUDA 10.2
Platform ubuntu18
ZED 3.3
Python 3.8
Downloading python package from https://download.stereolabs.com/zedsdk/3.3/ubuntu18/cu102/py38 ...

-> Please make sure numpy is installed 
 python3 -m pip install cython
 python3 -m pip install numpy
File saved into pyzed-3.3-cp38-cp38-linux_x86_64.whl
To install it run : 
 python3 -m pip install pyzed-3.3-cp38-cp38-linux_x86_64.whl
(base) adam@dual-18:/usr/local/zed$ python3 -m pip install pyzed-3.3-cp38-cp38-linux_x86_64.whl
Processing ./pyzed-3.3-cp38-cp38-linux_x86_64.whl
Installing collected packages: pyzed
Successfully installed pyzed-3.3
(base) adam@dual-18:/usr/local/zed$ cd samples/depth\ sensing/
(base) adam@dual-18:/usr/local/zed/samples/depth sensing$ cd python/
(base) adam@dual-18:/usr/local/zed/samples/depth sensing/python$ python depth_sensing.py 
Running Depth Sensing sample ... Press 'Esc' to quit
CAMERA STREAM FAILED TO START
...

[ I had to unplug the camera from the front USB 3.0 port to the rear USB 3.0 port--odd ]

...
(base) adam@dual-18:/usr/local/zed/samples/depth sensing/python$ python depth_sensing.py 
Running Depth Sensing sample ... Press 'Esc' to quit
[ZED][Init] No calibration file found for SN24925386. Downloading... 
[ZED][Init] Calibration file downloaded.
Traceback (most recent call last):
  File "depth_sensing.py", line 49, in <module>
    viewer = gl.GLViewer()
  File "/usr/local/zed/samples/depth sensing/python/ogl_viewer/viewer.py", line 218, in __init__
    self.camera = CameraGL()
  File "/usr/local/zed/samples/depth sensing/python/ogl_viewer/viewer.py", line 459, in __init__
    self.setDirection(tmp, tmp2)        
  File "/usr/local/zed/samples/depth sensing/python/ogl_viewer/viewer.py", line 513, in setDirection
    if(sl.Translation.dot_translation(self.vertical_, self.up_) < 0.):
AttributeError: type object 'pyzed.sl.Translation' has no attribute 'dot_translation'

```

mklingen-bd, 2020-11-17:
I have the same issue. The C++ api seems to define a function called `dot`, but it looks like the python API has no such function (or any function called `dot_translation`.

mklingen-bd, 2020-11-17:
specifically, I had to implement:

```
  37   │ def dot_translation(t1, t2):
  38   │     return np.dot(t1.get(), t2.get())
  39   │ 
```

Worth noting that while this made the example no longer crash, the example does not display anything but text (no opengl window is opened).

madasebrof, 2020-11-18:
On that note, the C++ version doesn't work either. It just fires up a blank Qt screen with a frustum/FOV widget:

[image]

It is connecting to the camera, as the blue LED lights up next to the ZED logo when I fire up `ZED_Depth_Sensing`

Running Ubuntu 18.04, CUDA 10.2

qt-truong, 2020-11-18:
Hi,

There was indeed an issue while trying to install the latest 3.3.1 release of the Python API. 
It has been fixed and you can now download it properly from our website (https://www.stereolabs.com/developers/release/, `SDK Downloads` section). This should resolve the error on the missing `dot_translation` function.

@mklingen-bd Is PyOpenGL installed on your PC ? 

@madasebrof Could you try running other samples from our repository using OpenGL to see if the same thing happens ?

madasebrof, 2020-11-18:
@qt-truong Looks like the SDK `ZED_SDK_Ubuntu18_full_cuda11.1_v3.3.1.run` hasn't changed from yesterday... (Same checksum)

Looks like the fix was made **pyzed** `pyzed-3.3-cp38-cp38-linux_x86_64.whl`, even though the build numbers didn't change.

I had to run:

```
python /usr/local/zed/get_python_api.py
python3 -m pip install --ignore-installed pyzed-3.3-cp38-cp38-linux_x86_64.whl
```

to overwrite the old pyzed wheel. 

Now the python depth sample is working well :) The C++ still doesn't work for depth (just shows an empty frame with a FOV widget).

ps, I had to make a slight modification to `get_python_api.py`, as it wasn't getting the CUDA version number properly. (Note: in CUDA 11.1, the file "/usr/local/cuda/version.txt" is no longer created.)

```
except FileNotFoundError:
        # Version.txt doesn't exist, check the var itself
        # temp = re.findall(r'\d+', cuda_path_version)
        # CUDA_MAJOR = int(temp[0])
        # CUDA_MINOR = int(temp[1])
        CUDA_MAJOR = "11"
        CUDA_MINOR = "1"
        print("CUDA " + str(CUDA_MAJOR) + "." + str(CUDA_MINOR))
        CUDA_STR = "cu" + str(CUDA_MAJOR) + str(CUDA_MINOR)
```

A better way to find the CUDA version on Linux is maybe running `nvcc --version`?

```
$ nvcc --version
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2020 NVIDIA Corporation
Built on Mon_Oct_12_20:09:46_PDT_2020
Cuda compilation tools, release 11.1, V11.1.105
Build cuda_11.1.TC455_06.29190527_0
```

Also, other samples ran fine, both python and C++ (e.g. Object Detection -> Birds Eye View)

adujardin, 2020-11-19:
@madasebrof Thank you, we'll make the appropriate changes in the setup scripts.

At the moment we don't have an explanation for the depth-sensing sample issue, we'll look into it. Is Depth Viewer (the tool) working correctly? It's very similar to this sample

Shishir-Kumar-Singh, 2021-01-09:
I am getting the same issue on the Jetson Xavier AGX device. 
On running the sample program it throws this error:
ZED/zed-examples-master/positional-tracking/python/positional_tracking.py 
Traceback (most recent call last):
  File "/home/shishir/ZED/zed-examples-master/positional-tracking/python/positional_tracking.py", line 50, in <module>
    viewer = gl.GLViewer()
  File "/home/shishir/ZED/zed-examples-master/positional-tracking/python/ogl_viewer/tracking_viewer.py", line 162, in __init__
    self.camera = CameraGL()
  File "/home/shishir/ZED/zed-examples-master/positional-tracking/python/ogl_viewer/tracking_viewer.py", line 476, in __init__
    self.setDirection(tmp, tmp2)
  File "/home/shishir/ZED/zed-examples-master/positional-tracking/python/ogl_viewer/tracking_viewer.py", line 528, in setDirection
    if(sl.Translation.dot_translation(self.vertical_, self.up_) < 0.):
AttributeError: type object 'pyzed.sl.Translation' has no attribute 'dot_translation'

Jetpack version is 4.4 and the sdk version is 3.3.3.
How to solve the error?

closed 2022-04-26 by github-actions[bot]