stereolabs/zed-sdk #309: get point3D and save as .pcd

adetails, 2020-11-19:
ZED 2 
Python3.7
when i use SVO recording/playback,i want to save a image and it's corresponding point_cloud which the point_cloud would be saved in a PCD subject.how can i do it?

adetails, 2020-11-19:
[image]
if i want to save the point_cloud in PCD as the same as the format the ZED Depth Viewer does.how can i do it

P-yver, 2021-12-14:
Hi,
The Python API let you save retreived data in different file format:
https://www.stereolabs.com/docs/api/python/classpyzed_1_1sl_1_1Mat.html#ae0568cd94ee09fa51eaf50679e3a2def

You can save images in jpg/png as well as Point Cloud in PCD/PLY

closed 2022-04-26 by github-actions[bot]