stereolabs/zed-sdk #312: SLAM realtime mesh volume restrictions

Tetsujinfr, 2020-11-19:
hi

is there a way to restrict in realtime the size of the mesh dynamically mapped with the SLAM algo?  e.g. restrict the mesh in memory for vertices withing X meters of the camera? or withing a cuboid of length x width x height meters?
Trying to retain only most important points (for me), reduce memory footprint and increase perf/stability overtime if possible.

thanks

P-yver, 2020-11-30:
Hi,
The current workflow does not support this kind of request, we will keep in mind your feedback for a further release.
For you, it would be like a constant memory/processing algorithm which delete data outside a certain range?

Tetsujinfr, 2020-11-30:
Yes exactlly. Either a box or sphere, or capsule hull. Ideally I could provide a hull mesh, and anything outside or inside the hull gets deleted dynamically to keep the overall mesh size under control. Helpful I think for robot navigation for instance. I am going to try to do this on CPU or GPU side, but I will only be able to touch tje final mesh of the SLAM while I think therr could be some memory benefit to do it at the SDK level.
Would be great if you can add it in a fiture realese.
Thanks for your reply.

closed 2022-04-26 by github-actions[bot]