stereolabs/zed-sdk #313: PCD save issue

adetails, 2020-11-20:
How can i save the value get from (point3D = point_cloud.get_value(i, j) to a pcd file ,which it's format is the same as i use the ZDE Depth Viewer?
Besides,if i use the follow code,the format of the value of XYZ are not the same as using the ZDE Depth Viewer.
cam = sl.Camera()
cam.retrieve_measure(point_cloud, sl.MEASURE.XYZRGBA)
pcd_data = point_cloud.write(filepath1)

the follow is one value of a XYZRGB in pcd file :
ZDE Depth Viewer : [-11.988397 6.658359 -10.071815 5.348763e-39]

by my code : [-9730.087891 -5482.849121 8293.923828 4.703042e-39]

adetails, 2020-11-20:
@P-yver  Excuse me, can you help me with my problem?Thank you very much

Myzhar, 2020-11-20:
@adetails have you changed the coordinate units to meter instead of using the default?
See here:
https://www.stereolabs.com/docs/api/structsl_1_1InitParameters.html#a75cb1e715c1be19c591d21fa2591b421

adetails, 2020-11-20:
@Myzhar Thanks.
I am not sure if i have changed the coordinate units,i just used the reav_svo.py git from the API and i just dded a line of code(pcd_data = point_cloud.write(filepath1)).
How can i check if i have changed the coordinate units?

adetails, 2020-11-20:
@Myzhar Do you how to correctly save the Point Cloud into PCD files as same as the ZDE Depth Viewer saved?Thank you so much.

adetails, 2020-11-21:
@Myzhar  I tried to set the  coordinate units as below.But it was wrong .
input_type = sl.InputType()
input_type.set_from_svo_file(filepath)
init = sl.InitParameters(input_t=input_type, svo_real_time_mode=False, coordinate_units=UNIT.MILLIMETER )

The type of error is :
NameError: name 'UNIT' is not defined

P-yver, 2020-11-23:
You have to change the sl::UNIT to METERS and use the `sl::COORDINATE_SYSTEM::RIGHT_HANDED_Y_UP`
``` cpp
    InitParameters init_parameters;
    init_parameters.coordinate_units = sl::UNIT::METER; // set metric in meter
    init_parameters.coordinate_system = sl::COORDINATE_SYSTEM::RIGHT_HANDED_Y_UP; // OpenGL's coordinate 
  ```

closed 2022-04-26 by github-actions[bot]