stereolabs/zed-sdk #315: Can't get reduce point cloud resolution with retrieveMeasure?

niwhsa9, 2020-11-30:
I am able to get point clouds with the full camera resolution 720p, but when I try to get pointclouds of 320x180, I get "Error (11) Invalid Argument". Not sure why this is happening: 

```
sl::Resolution cloud_res(320, 180);
sl::Mat gpu_cloud (cloud_res, sl::MAT_TYPE::F32_C4, sl::MEM::GPU);
...
while(...) {
        zed.grab();
        zed.retrieveMeasure(gpu_cloud, sl::MEASURE::XYZRGBA, sl::MEM::GPU, cloud_res); 
}
```

I've also tried modifying the [point cloud depth sensing example code](https://github.com/stereolabs/zed-examples/blob/master/depth%20sensing/cpp/src/main.cpp) by changing:
**line 71**
```
Resolution res(320, 180);
Mat point_cloud(res, MAT_TYPE::F32_C4, MEM::GPU);

```
**line 78**
```
zed.retrieveMeasure(point_cloud, MEASURE::XYZRGBA, MEM::GPU, res);
```
and experienced the same error

P-yver, 2020-11-30:
Hi, 
You must update the sample this way to make it works as expected:

```` cpp
sl::Resolution cloud_res(320, 180);
auto camera_config = zed.getCameraInformation(cloud_res).camera_configuration;

Mat point_cloud(camera_config.resolution, MAT_TYPE::F32_C4, MEM::GPU);
// 
zed.retrieveMeasure(point_cloud, MEASURE::XYZRGBA, MEM::GPU, cloud_res);
````

in this case `camera_config.resolution` is set to `cloud_res`.
`GLViewer::init` must be called with this value too.

niwhsa9, 2020-12-01:
You were absolutely right, thank you!

closed 2020-12-01 by niwhsa9