stereolabs/zed-sdk #317: Examples issues on Xavier NX

IThraen, 2020-12-01:
Hi,

I just installed the ZED SDK 3.3.2 for jetpack 4.4 on a Xavier NX, and some examples are not working properly.
For example, body tracking throws an out of memory error (cuda related), and the image viewer (object detection) throws a segfault.
I tried installing both Jetpack 4.4 and 4.4.1.

Is there something i missed?

Tetsujinfr, 2020-12-11:
pls share your error details (log). Also are you trying the cpp examples or the pyhton version? which exemples specifically do not work?

Globibluk, 2021-01-05:
Sorry for the long delay.
I'm running the cpp versions of body tracking and object detection (image viewer).
Here are the logs:

Body tracking:

> Optimized model not found, trying to download it...
> 
> [ZED][Object Detection] AI model not found, downloading http://download.stereolabs.com/ai_model/skeleton_performance_1_0
>   |/-\ Donein void ai::BatchedBuffer::allocate(size_t, bool, bool) : cuda error [1]: invalid argument.

Object detection:

> Optimized model not found, trying to download it...
> 
> [ZED][Object Detection] AI model not found, downloading http://download.stereolabs.com/ai_model/objects_performance_2_0
>   |/-\ DoneSegmentation fault (core dumped)

Tetsujinfr, 2021-01-05:
I assume you have a ZED2 camera, but my ZED mini camera does not support those 2 examples, I am afraid I cannot help you much on those issues.

From experience, for the object detection, when you have a segmentation fault this could be a memory issue (not enough memory). You can monitor the memory load of the NX with jtop while running the exemple, and check if it goes through the roof. If you have a SSD you can setup a large swap file so that the memory issue can be eventually managed (it will run but it will run very slowly most likely).

For the body tracking, looks like there is some cofe issue or some incorrect parameters passed.
Sorry for not being able to help more.

Hopefully someone from Stereolabs will step in on this at some point.

closed 2022-04-26 by github-actions[bot]

interqhq, 2022-07-14:
> Hi,
>
> I just installed the ZED SDK 3.3.2 for jetpack 4.4 on a Xavier NX, and some examples are not working properly. For example, body tracking throws an out of memory error (cuda related), and the [...]

hi,
I have got the same situtation, and my platform is Jetson AGX Xavier, do you have resolved the problem?
thanks