stereolabs/zed-sdk #319: OpenGL.error.NullFunctionError

hyungju-song, 2020-12-02:
I use **zed2, python 3.7.7 , win10**
I was able to run almost example code, except code using **opengl**.
 (ex. i can run:  camera_control 
        i can't run: body_tracking, positional_tracking)
After several trials and errors, I realized that all the code related to **ogl_viewer** showed the same error.

After seeing the advice that the lib called freeglut.dll may not exist, I installed freeglut 3.0 newly, but I confirmed that there is freeglut2.8 which is automatically installed in zed-sdk. Also, there was freeglut2.6 inside while laying cuda10.0.
I installed the sdk for cuda10.0-win10, but it is a bit strange that the versions of freeglut are different. Isn't this the original problem?
 
Is there no error like me?

The error is as follows:
**OpenGL.error.NullFunctionError: Attempt to call an undefined function glutInit, check for bool(glutInit) before calling**

HELP ME OUT , PLZ
Thanks in advance.

**my Software/Hardware specs:** 
win10
geforce mx150
python 3.7.7 (tags/v3.7.7:d7c567b08f, Mar 10 2020, 10:41:24) [MSC v.1900 64 bit (AMD64)] on win32
CUDA 10.0 
pycharm community 2020

P-yver, 2020-12-02:
Hi,
Indeed, on windows you may encouter this issue related to opengl.
You may update your version, you can find the latest ones [here](https://www.lfd.uci.edu/~gohlke/pythonlibs/#pyopengl)
In your case (Python 3.7) you will need to download and install:
 - PyOpenGL‑3.1.5‑cp37‑cp37m‑win_amd64.whl
-  PyOpenGL_accelerate‑3.1.5‑cp37‑cp37m‑win_amd64.whl

to install them
> python3 -m pip install *PyOpenGL_XXXX.whl*

hyungju-song, 2020-12-03:
Thank you. 
Surprisingly, it was resolved immediately.
I can now run all my code 💯

Kazlovich, 2021-01-15:
Hello! I am running into a very similar issue. When I run the body_tracking tutorial, I keep getting the ModuleNotFoundError: No module named 'ogl_viewer'

I have downloaded the PyOpenGl extensions as suggested and still get the error. 

I have ZED2, Windows 10, Python 3.7.8 64-bit version. 
Please help!

qt-truong, 2021-01-22:
Hi, 

This error usually means that the script you ran cannot find the folder `ogl_viewer` in which OpenGL related code is under.
If you copied the sample `positional_tracking.py` locally on your computer, make sure to also copy everything that is under `positional tracking/python/`
The folder in which you run the sample should look like :
```bash
<your_folder_name>
├── ogl_viewer/
└── positional_tracking.py
```

KastanDay, 2021-07-07:
@qt-truong had the correct solution for me. You can find those files here (but I recommend just cloning the ENTIRE repo, much easier): https://github.com/stereolabs/zed-examples/tree/master/object%20detection/image%20viewer

closed 2022-04-26 by github-actions[bot]