stereolabs/zed-sdk #320: Camera Capture Timestamp Questions

lubertRoft, 2020-12-04:
[image]

According to the description, the camera clock can be synchronized with the host clock through PTP. How can it be done? The camera is connected to the host by USB, how can I do PTP synchronization? Or does it mean the connection between different host computers? Or is it also possible to connect the camera to the host computer via Ethernet?
 
With the ZED SDK can we use gettimestamp(). What is the reference time of this function? Is that from Host or from Sensor? Is this time equal to the time when the image was made or equal to the time when the image was arrived at the host?
 
I saw that the Latency of the image to arrive to the host when using a frame rate of 100 is between 20 and 30 msec. Is this a 100% sure number or can it vary depending on the use case? If i have a Real Time Operating System and a good USB connection will the time decrease?
 
Thank you for your help

Myzhar, 2020-12-04:
Hi @lubertRoft 
PTP and NTP can be used to synchronize multiple cameras connected to different hosts in the same local network.
When you use the function `sl::zed::getTimestamp()` you can retrieve two different types of information:
* When requesting the `TIME_REFERENCE::IMAGE` timestamp, the value corresponds to the time at which the entire image was available in the PC memory. As such, it ignores the communication time that corresponds to 2 or 3 frame-time based on the fps (ex: 33.3ms to 50ms at 60fps).
* When requesting the `TIME_REFERENCE::CURRENT` timestamp, the current nanosecond timestamp is returned.

The latency highly depends on the operating system and on the type required by it to handle the USB communication, but also on the resolution and the framerate used.

More info:
C++: https://www.stereolabs.com/docs/api/classsl_1_1Camera.html#a3cd31c58aba33727f35aeae28244c82d
Python: https://www.stereolabs.com/docs/api/python/classpyzed_1_1sl_1_1Camera.html#a5a84f1222b1968b0b05e10241763f050

lubertRoft, 2020-12-04:
Alright, thank you very much for the fast reply!

closed 2020-12-04 by lubertRoft