stereolabs/zed-sdk #322: spatial mapping result was very bad

hudmgy, 2020-12-13:
I used the zed2 with tools of ZEDfu from sdk3.3.3 to get a spatial mapping of a room, but the results were terrible. There are a lot of holes and disconnections in it. Did I do something wrong？

obraun-sl, 2020-12-15:
Hi,
You can try to record a SVO file (with ZED Explorer) and then process it in ZEDFu to see if you have differences. If yes, then maybe the compute power of your PC is not enough to handle live mapping properly.

hudmgy, 2020-12-16:
> Hi,
> You can try to record a SVO file (with ZED Explorer) and then process it in ZEDFu to see if you have differences. If yes, then maybe the compute power of your PC is not enough to handle live mapping properly.

Hi
Thank you for reply！The result is better when using SVO file as input.
My computer is Jetson Xavier NX,  is it not enough?
What types of computer do you recommend?

thank you

obraun-sl, 2020-12-28:
Xavier AGX will give better performances, even if Xavier NX should be enough. Make sure you are in max performance mode (MAXN).

Tetsujinfr, 2021-01-10:
I have pretty good results on NX, fyi.
What resolution did you run SLAM on? Are you sure GPU is used?

closed 2022-04-26 by github-actions[bot]