stereolabs/zed-sdk #323: SVO Playback

oscarthorn, 2020-12-15:
Hi!

Is it possible to make the SVO playback behavior identical to using a camera? What I specifically am looking for is, if I'm using a camera and request say 60fps but my hardware is to slow to actually call grab often enough I might end up with 40fps and the rest of the frames get dropped. 

However if I use a SVO file as input each frame gets processed no matter what. Leading to slow playback. As I mainly want to use the SVO files for debugging and development I want to get as close to having a camera plugged in as possible.

obraun-sl, 2020-12-15:
Hi,
You can set the InitParameters::svo_real_time_mode to true to have a behavior similar to real time mode. It should drop frames if needed.

oscarthorn, 2020-12-15:
Great, thanks!

closed 2020-12-15 by oscarthorn