stereolabs/zed-sdk #326: Compile error when I  compiling zed-example

1556260456, 2020-12-26:
After I installed zed-sdk and cuda,I download zed-examples,and I create a Build document ,then I run:cmake ..
I got this error:
adam@C:~/ZED2/zed-examples/build$ cmake ..
-- A library with BLAS API found.
-- A library with BLAS API found.
-- A library with BLAS API found.
-- Found CUDA: /usr/local/cuda (found version "11.1") 
-- A library with BLAS API found.
-- Found CUDA: /usr/local/cuda (found suitable version "11.1", minimum required is "11.1") 
-- Found CUDA: /usr/local/cuda (found version "11.1") 
-- A library with BLAS API found.
-- Found CUDA: /usr/local/cuda (found suitable version "11.1", minimum required is "11.1") 
-- Found CUDA: /usr/local/cuda (found version "11.1") 
-- A library with BLAS API found.
-- Found CUDA: /usr/local/cuda (found suitable version "11.1", minimum required is "11.1") 
-- A library with BLAS API found.
-- A library with BLAS API found.
-- A library with BLAS API found.
-- A library with BLAS API found.
-- A library with BLAS API found.
-- A library with BLAS API found.
-- A library with BLAS API found.
-- A library with BLAS API found.
-- A library with BLAS API found.
-- A library with BLAS API found.
-- A library with BLAS API found.
-- A library with BLAS API found.
-- A library with BLAS API found.
-- A library with BLAS API found.
-- A library with BLAS API found.
-- A library with BLAS API found.
-- A library with BLAS API found.
-- A library with BLAS API found.
CMake Error at CMakeLists.txt:36 (INSTALL):
  install TARGETS given target "ZED_Camera_Control" which does not exist in
  this directory.

-- Configuring incomplete, errors occurred!
See also "/home/adam/ZED2/zed-examples/build/CMakeFiles/CMakeOutput.log".
See also "/home/adam/ZED2/zed-examples/build/CMakeFiles/CMakeError.log".
  How could I fix this error,did I forget to install any software?

obraun-sl, 2020-12-28:
Hi,
What is your version of cmake ? ($ cmake --version)
This error might come from a old version of cmake (Did not reproduce with cmake 3.16). 
Either update cmake or build each sample separately.

1556260456, 2020-12-31:
Hi，My cmake version is :
[image]
And I can build each sample separately.
I have another proplem:
first, after I build the zed-ros-wrapper,I run  :roslaunch zed.launch
I got this error:
[image]
[image]

[image]
[image]
 I have installed the ZED_SDK_Ubuntu16_cuda11.1_v3.3.3.run,Can you tell me how to fix this error?

obraun-sl, 2020-12-31:
Hi,
It seems that there is a mismatch between the GPU driver and the CUDA version. 
What is the output of 
$ nvidia-smi ?

1556260456, 2020-12-31:
Thank you for replying so fast.
It seem that the error has been fixed after I installed a another Driver ,And now I can run this demo.
anyway,the output after run :nvidia-smi  **### is** 

+-----------------------------------------------------------------------------+
| NVIDIA-SMI 455.23.05    Driver Version: 455.23.05    CUDA Version: 11.1     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  GeForce GTX 1050    Off  | 00000000:01:00.0  On |                  N/A |
| N/A   44C    P0    N/A /  N/A |    386MiB /  2000MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A      1115      G   /usr/lib/xorg/Xorg                363MiB |
|    0   N/A  N/A      2029      G   compiz                             20MiB |
+-----------------------------------------------------------------------------+

closed 2021-01-02 by 1556260456