stereolabs/zed-sdk #329: Depth Sensing issue on Jetson Xavier

Brett-Wildermoth, 2021-01-06:
When conducting a Depth Scan, I only appear to get half of the scan data. It appears I only get the data from the left camera and not the data for both cameras.

Below is my relevant code snippet

```
....
if(mzed.grab(m_rt) == sl::ERRORCODE::SUCCESS){
   bool success = mzed.retrieveMeasure(m_datacloud, sl::MEASURE::XYZRGBA) == sl::ERRORCODE::SUCCESS;
   for (int rowIdx=0; rowIdx < m_datacloud.getWidth(); rowIdx++){
     for(int colIdx = 0; colIdx < m_datacloud.getHeight(); colIdx++){
        sl::float4 pointcloudvalue;
        m_datacloud.getValue(colIdx, rowIdx, &pointcloudvalue);
       if(isfinite(pointcloudvalue.z){
        m_surfaceBuffer[rowIdx][colIdx].x = pointcloudvalue.x;
        m_surfaceBuffer[rowIdx][colIdx].y = pointcloudvalue.y;
        m_surfaceBuffer[rowIdx][colIdx].z = pointcloudvalue.z;
       }
…
     }
  }
}
....
```

Swapping colIdx and rowIdx in mdatacloud.getValue() makes no difference.

m_rt is

`m_rt.sensingmode = sl::SENSINGMODE::STANDARD;
`
and the initparameters for the ZED are

```
minitparams.cameraresolution = sl::RESOLUTION::HD1080;
minitparams.camerafps = 10;
minitparams.coordinateunits = sl::UNIT::METER;
minitparams.depthminimumdistance = 0.05 ;
minitparams.coordinatesystem = sl::COORDINATESYSTEM::IMAGE;
minitparams.depthmode = sl::DEPTHMODE::ULTRA ;
```
I have setup my code similar to the examples and the demo code.

Doing a m_datacloud.write() seems to produce different data?

Brett-Wildermoth, 2021-01-06:
Also JetPack version is 4.2 with Cuda 10 and the Zed Library version is 3.22.

P-yver, 2021-01-06:
Hi,

In the ZED SDK the reference is the left camera, every data are given according to it (depth, point cloud, positional tracking).
If you want the data of the right camera you can enable it (it will require more computation).

    // initparameters 
    minitparams.enable_right_side_measure = true;
   
    // then, retrieve the right data
    sl::Mat pc_right;
    zed.retrieveMeasure(pc_right, sl::MEASURE::XYZRGBA_RIGHT);

Brett-Wildermoth, 2021-01-06:
Using the right camera gave me the same data. Based on what I have read when I use the left camera I should get a depth value that kind of aligns with the image I get out of the left camera. What I get appears to be only just over half of what I see in the image.

Brett-Wildermoth, 2021-01-06:
Major problem I found was I had not initialized the point_cloud properly. I appear to be getting the correct range now. Post processing is now clipping it and I will have to source the issue there.

closed 2021-01-11 by Brett-Wildermoth