stereolabs/zed-sdk #330: Freezing Problem while tracking

Baldins, 2021-01-07:
Hi, 

Since the new SDK has been released I am having some camera freezing problem while tracking pedestrians outdoor. 
I am not sure of the reason why this is happening. I have been trying to figure this out for weeks now, but it still keep freezing after few minutes. 

Does anyone have a solution or an explanation for this?

Thank you in advance for your reply! 

Francesca

obraun-sl, 2021-01-08:
Is the data frozen or the camera? For example, does the grab() function stop returning SUCCESS?
What is the version of the ZED SDK you are using? Does it freeze also with previous version?

Baldins, 2021-01-08:
Yes, everything freezes. 
The camera is not streaming anymore and same things for the tracking data I am printing out on the terminal. 

Do you have any idea of what is causing this problem? 

Thank you!

edap, 2021-01-22:
I have the same problem too while tracking bodies. The example application freezes after a while. It happens at random interval, but you can reproduce the error moving the camera with your hand while tracking. I was not able to copy and paste the output, therefore I have taken a screenshot.

[image]

edap, 2021-02-19:
Any news on this front?

Myzhar, 2021-02-19:
@Baldins @edap can you provide more information about the configuration of the host machine (GPU, OS, CUDA version)?
It's also important to know the parameters that you are using for camera initialization, runtime elaboration, and object detection.
Thank you

edap, 2021-02-22:
Hello @Myzhar , this is the output of the zed diagnostic tool:

```
{
    "Camera Test": {
        "ZEDCount": 1,
        "deviceInitResult": 0,
        "firmware": 1523,
        "resolutions": {
            "HD1080@15": {
                "fps (input)": 15,
                "fps (output)": 15,
                "height (input)": 1080,
                "height (output)": 1080,
                "initCode": true,
                "initialization": "Ok",
                "width (input)": 1920,
                "width (output)": 1920
            },
            "HD1080@30": {
                "fps (input)": 30,
                "fps (output)": 27,
                "height (input)": 1080,
                "height (output)": 1080,
                "initCode": true,
                "initialization": "Ok",
                "width (input)": 1920,
                "width (output)": 1920
            },
            "HD2K@15": {
                "fps (input)": 15,
                "fps (output)": 15,
                "height (input)": 1242,
                "height (output)": 1242,
                "initCode": true,
                "initialization": "Ok",
                "width (input)": 2208,
                "width (output)": 2208
            },
            "HD720@15": {
                "fps (input)": 15,
                "fps (output)": 15,
                "height (input)": 720,
                "height (output)": 720,
                "initCode": true,
                "initialization": "Ok",
                "width (input)": 1280,
                "width (output)": 1280
            },
            "HD720@30": {
                "fps (input)": 30,
                "fps (output)": 30,
                "height (input)": 720,
                "height (output)": 720,
                "initCode": true,
                "initialization": "Ok",
                "width (input)": 1280,
                "width (output)": 1280
            },
            "HD720@60": {
                "dirty lenses": 0,
                "dropCount": 0,
                "fps (input)": 60,
                "fps (output)": 49,
                "height (input)": 720,
                "height (output)": 720,
                "imageCount": 501,
                "initCode": true,
                "initialization": "Ok",
                "tearingCount": 0,
[... 230 lines cut ...]
                "PhysicalName": "\\Device\\NTPNP_PCI0002"
            }
        ],
        "valid": [
            "<b>USB Bandwidth: </b> OK"
        ]
    },
    "ZED SDK": {
        "CUDA Toolkit version": "V11.1.105",
        "CUDA_reachable": true,
        "DiagnosticZEDSDKVersion": "3.3.3",
        "InstalledZEDSDKVersion": "3.3.3",
        "ZEDSDKFound": true,
        "driverVersion": "456.81",
        "drivervDllVersion": "27.21.14.5681",
        "initResult": 1,
        "loaded": true,
        "runtimeDllVersion": 11010,
        "valid": [
            "<b>ZED SDK version:</b> 3.3.3"
        ]
    }
}

```

Myzhar, 2021-02-23:
@edap please post also the configuration that you are using inside your SW: `InitParameters`, `RuntineParameters`, `ObjectDetectionParameters`, `ObjectDetectionRuntimeParameters`.
It's important for us to replicate your problem and to find the cause of the freezing/crash

edap, 2021-02-23:
Thanks for your support @Myzhar . This is the code that I am running https://github.com/edap/zed-examples/blob/master/zedTop/cpp/src/TOP_main.cpp. It is a custom operator for Touch Designer.
It is based on this example https://github.com/stereolabs/zed-examples/blob/master/body%20tracking/cpp/src/main.cpp

These are the parameters that I am using:

**Init parameters:**

```
init_parameters.camera_resolution = RESOLUTION::HD2K;
init_parameters.depth_mode = DEPTH_MODE::ULTRA;
init_parameters.coordinate_system = COORDINATE_SYSTEM::RIGHT_HANDED_Y_UP;
init_parameters.coordinate_units = UNIT::METER;
```

**runtime parameters:**

None.

**object detection parameters:**

```
obj_det_params.enable_tracking = true; // track people across images flow
obj_det_params.enable_body_fitting = true; // smooth skeletons moves
obj_det_params.detection_model = DETECTION_MODEL::HUMAN_BODY_ACCURATE;
```

**object detection runtime parameters:**

```
objectTracker_parameters_rt.detection_confidence_threshold = 50;
```

edap, 2021-03-31:
Hello everybody, any News on this front?

closed 2022-04-26 by github-actions[bot]