stereolabs/zed-sdk #331: CUDA related problem running simple  hello example

Shishir-Kumar-Singh, 2021-01-09:
i am getting the following issue with the ZED2 camera sample tutorial python code running:
python3 ~/zed-examples-master/tutorials/tutorial1-hello-ZED/python/hello_zed.py
CUDA error at Camera.cpp:81 code=101(cudaErrorInvalidDevice) "void sl::Camera::close()" 
CUDA error at Camera.cpp:81 code=101(cudaErrorInvalidDevice) "void sl::Camera::close()" 
CUDA error at Camera.cpp:81 code=101(cudaErrorInvalidDevice) "void sl::Camera::close()" 
CUDA error at Camera.cpp:135 code=101(cudaErrorInvalidDevice) "void sl::Camera::close()" 
CUDA error at Camera.cpp:169 code=101(cudaErrorInvalidDevice) "void sl::Camera::close()" 
CUDA error at CameraUtils.hpp:648 code=101(cudaErrorInvalidDevice) "void sl::ObjectsDetectorHandler::clear()" 
CUDA error at CameraUtils.hpp:662 code=101(cudaErrorInvalidDevice) "void sl::ObjectsDetectorHandler::clear()" 
CUDA error at CameraUtils.hpp:668 code=101(cudaErrorInvalidDevice) "void sl::ObjectsDetectorHandler::clear()" 
CUDA error at CameraUtils.hpp:671 code=101(cudaErrorInvalidDevice) "void sl::ObjectsDetectorHandler::clear()" 
CUDA error at CameraUtils.hpp:675 code=101(cudaErrorInvalidDevice) "void sl::ObjectsDetectorHandler::clear()" 
CUDA error at CameraUtils.hpp:682 code=101(cudaErrorInvalidDevice) "void sl::ObjectsDetectorHandler::clear()" 
CUDA error at Camera.cpp:184 code=101(cudaErrorInvalidDevice) "void sl::Camera::close()"

OS: Ubuntu 18.04
ZED SDK: 3.3.3
CUDA: 10.2
Nvidia driver version: 450.57 

what could be the issue?

obraun-sl, 2021-01-11:
Can you change the sdk_verbose to true here : 
https://github.com/stereolabs/zed-examples/blob/master/tutorials/tutorial%201%20-%20hello%20ZED/python/hello_zed.py#L30

to have more output about what's wrong.

Shishir-Kumar-Singh, 2021-01-13:
> Can you change the sdk_verbose to true here :
> https://github.com/stereolabs/zed-examples/blob/master/tutorials/tutorial%201%20-%20hello%20ZED/python/hello_zed.py#L30
> 
> to have more output about what's wrong.

After changing this i am getting following info:
CUDA error at Camera.cpp:81 code=101(cudaErrorInvalidDevice) "void sl::Camera::close()" 
CUDA error at Camera.cpp:81 code=101(cudaErrorInvalidDevice) "void sl::Camera::close()" 
**No NVIDIA graphics card detected. Install an NVIDIA GPU, CUDA and restart your computer after completing installation.**
CUDA error at Camera.cpp:81 code=101(cudaErrorInvalidDevice) "void sl::Camera::close()" 
CUDA error at Camera.cpp:135 code=101(cudaErrorInvalidDevice) "void sl::Camera::close()" 
CUDA error at Camera.cpp:169 code=101(cudaErrorInvalidDevice) "void sl::Camera::close()" 
CUDA error at CameraUtils.hpp:648 code=101(cudaErrorInvalidDevice) "void sl::ObjectsDetectorHandler::clear()" 
CUDA error at CameraUtils.hpp:662 code=101(cudaErrorInvalidDevice) "void sl::ObjectsDetectorHandler::clear()" 
CUDA error at CameraUtils.hpp:668 code=101(cudaErrorInvalidDevice) "void sl::ObjectsDetectorHandler::clear()" 
CUDA error at CameraUtils.hpp:671 code=101(cudaErrorInvalidDevice) "void sl::ObjectsDetectorHandler::clear()" 
CUDA error at CameraUtils.hpp:675 code=101(cudaErrorInvalidDevice) "void sl::ObjectsDetectorHandler::clear()" 
CUDA error at CameraUtils.hpp:682 code=101(cudaErrorInvalidDevice) "void sl::ObjectsDetectorHandler::clear()" 
CUDA error at Camera.cpp:184 code=101(cudaErrorInvalidDevice) "void sl::Camera::close()"

But i have a working card.

obraun-sl, 2021-01-13:
Can you type in a terminal : 
$ nvidia-smi 
and give me the output ?

Shishir-Kumar-Singh, 2021-01-14:
> Can you type in a terminal :
> $ nvidia-smi
> and give me the output ?

following is the output:

Thu Jan 14 22:13:40 2021       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 450.57       Driver Version: 450.57       CUDA Version: 11.0     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  GeForce GTX 107...  Off  | 00000000:01:00.0  On |                  N/A |
| N/A   49C    P0    32W /  N/A |    310MiB /  8116MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A      2133      G   /usr/lib/xorg/Xorg                130MiB |
|    0   N/A  N/A      7320      G   /usr/bin/gnome-shell               66MiB |
|    0   N/A  N/A      7690      G   /opt/freedownloadmanager/fdm        1MiB |
|    0   N/A  N/A      9011      G   ...AAAAAAAAA= --shared-files      107MiB |
+-----------------------------------------------------------------------------+

Shishir-Kumar-Singh, 2021-01-14:
[image]

obraun-sl, 2021-01-15:
* Did you try to reboot your PC ? It seems the GPU is not "loaded". 
* You can try 3.4.0RC to see if it solves your problem --> https://community.stereolabs.com/t/zed-sdk-3-4-0-release-candidate/58

Shishir-Kumar-Singh, 2021-01-16:
> * Did you try to reboot your PC ? It seems the GPU is not "loaded".
> * You can try 3.4.0RC to see if it solves your problem --> https://community.stereolabs.com/t/zed-sdk-3-4-0-release-candidate/58

I did try to reboot the pc but no avail.
I also tried the 3.4.0 RC version that one is currently installed in the system but it did not solve the problem.
I use my Nvidia card with other cuda related framework like tensorflow etc and it works with them.
for the information i have Nvidia GTX 1070 card.

closed 2022-04-26 by github-actions[bot]