stereolabs/zed-sdk #333: How can i uninstall SDK 3.3.3 completely?

gaizerik, 2021-01-13:
Hi, I have problem with SDK version 3.3.3.

It seems my problem is similar with #331.
I have some network with ZED2 HD resolution input.
And it generates error as below 
=========================================
RuntimeError: CUDA error: an illegal memory access was encountered
python-BaseException
THCudaCheck FAIL file=/pytorch/aten/src/THC/THCCachingHostAllocator.cpp line=278 error=700 : an illegal memory access was encountered
CUDA error at Camera.cpp:81 code=77(cudaErrorIllegalAddress) "void sl::Camera::close()" 
CUDA error at /builds/sl/ZEDKit/lib/src/sl_zed/Camera.cpp:88 code=77(cudaErrorIllegalAddress) "cudaStreamSynchronize(h->global_stream)" 
CUDA error at Camera.cpp:101 code=77(cudaErrorIllegalAddress) "void sl::Camera::close()" 
[ZED][Mat] [WARNING] n/a Mat with GPU memory allocated should be freed before deleting the Camera object to avoid GPU context issues.
[ZED][Mat] [WARNING] n/a Mat with GPU memory allocated should be freed before deleting the Camera object to avoid GPU context issues.
[ZED][Mat] [WARNING] n/a Mat with GPU memory allocated should be freed before deleting the Camera object to avoid GPU context issues.
[ZED][Mat] [WARNING] n/a Mat with GPU memory allocated should be freed before deleting the Camera object to avoid GPU context issues.
[ZED][Mat] [WARNING] n/a Mat with GPU memory allocated should be freed before deleting the Camera object to avoid GPU context issues.
[ZED][Mat] [WARNING] n/a Mat with GPU memory allocated should be freed before deleting the Camera object to avoid GPU context issues.
[ZED][Mat] [WARNING] n/a Mat with GPU memory allocated should be freed before deleting the Camera object to avoid GPU context issues.
[ZED][Mat] [WARNING] n/a Mat with GPU memory allocated should be freed before deleting the Camera object to avoid GPU context issues.
[ZED][Mat] [WARNING] n/a Mat with GPU memory allocated should be freed before deleting the Camera object to avoid GPU context issues.
[ZED][Mat] [WARNING] n/a Mat with GPU memory allocated should be freed before deleting the Camera object to avoid GPU context issues.
in void sl::Mat::free(sl::MEM) : cuda error [77]: an illegal memory access was encountered.
in void sl::Mat::free(sl::MEM) : cuda error [77]: an illegal memory access was encountered.
in void sl::Mat::free(sl::MEM) : cuda error [77]: an illegal memory access was encountered.
in void sl::Mat::free(sl::MEM) : cuda error [77]: an illegal memory access was encountered.
...............
==========================================

When I was using  version of 3.2.2, there was no problem.
So I want to roll-back SDK to 3.2.2 version, because i have no time to solve this problem with 3.3.3.
I cannot find any instruction to uninstall 3.3.3. 
Could you plz let me know how to do that?

Regards,
SYLee.

obraun-sl, 2021-01-13:
If you are on Linux : 
* Remove /usr/local/zed --> $ sudo rm -R /usr/local/zed/
* Install the ZED SDK 3.2.2

gaizerik, 2021-01-13:
@obraun-sl 
I've already tried it (remove /usr/local/zed and install 3.2.2).
Then i encountered below error.
......................................................................
   import pyzed.sl as sl
ImportError: /home/sylee81/anaconda3/envs/cu10/lib/python3.6/site-packages/pyzed/sl.cpython-36m-x86_64-linux-gnu.so: undefined symbol: _ZN2sl14InitParametersC1ENS_10RESOLUTIONEibNS_10DEPTH_MODEENS_4UNITENS_17COORDINATE_SYSTEMEbiffbibNS_6StringEiP8CUctx_stNS_9InputTypeES5_bb
......................................................................

adujardin, 2021-01-14:
@gaizerik you can either manually remove the package by deleting the folder `/home/sylee81/anaconda3/envs/cu10/lib/python3.6/site-packages/pyzed`

Or preferably reinstall the version you want using `--ignore-installed` with pip, like : 
```
python3 -m pip install pyzed-3.2-cp36-cp36m-linux_x86_64.whl --ignore-installed
```
It will overwrite the package.

closed 2022-04-26 by github-actions[bot]