stereolabs/zed-sdk #337: IMU pose of SensorData when using streaming

phamdat09, 2021-01-18:
Hello,

I am a bit confused about value of imu pose when I use `getSensorsData(mSensorData, sl::TIME_REFERENCE::IMAGE)` function in the streaming. These values of sender and reciever are not the same. As I understood, two values should be the same. Am I correct or wrong ? Thanks

**My test**
I only use the code of sender and reciver in the sample folder and just add this line to show the value of IMU pose:
`if (zed.getSensorsData(mSensorData, sl::TIME_REFERENCE::IMAGE) == sl::ERROR_CODE::SUCCESS){
            print("mImuPose value: ", sl::ERROR_CODE::SUCCESS,  mSensorData.imu.pose.getInfos().c_str());
        }`

one more thing, i change the  `init_parameters.coordinate_system = COORDINATE_SYSTEM::RIGHT_HANDED_Y_UP;` to `init_parameters.coordinate_system = COORDINATE_SYSTEM::IMAGE;` in the reciver to make sur that both of them have the same coordinate
**Code of sender**

#include <stdio.h>
#include <string.h>

// ZED includes
#include <sl/Camera.hpp>

// Sample includes
#include "utils.hpp"

// Using namespace
using namespace sl;
using namespace std;

void print(string msg_prefix, ERROR_CODE err_code = ERROR_CODE::SUCCESS, string msg_suffix = "");
int parseArgs(int argc, char **argv, sl::InitParameters& param);

int main(int argc, char **argv) {
    // Create a ZED camera
    Camera zed;

    // Set configuration parameters for the ZED
    InitParameters init_parameters;
    init_parameters.camera_resolution = sl::RESOLUTION::HD720;
    init_parameters.depth_mode = DEPTH_MODE::NONE;
    init_parameters.sdk_verbose = true;
    init_parameters.camera_fps = 30;
    init_parameters.input.setFromCameraID(1);
    int res_arg = parseArgs(argc, argv, init_parameters);

    // Open the camera
    auto returned_state = zed.open(init_parameters);
    if (returned_state != ERROR_CODE::SUCCESS) {
        print("Camera Open", returned_state, "Exit program.");
        return EXIT_FAILURE;
    }

    StreamingParameters stream_params;
    stream_params.codec = STREAMING_CODEC::H265;
    stream_params.bitrate = 8000;
    // stream_params.chunk_size = 4096;
    if (argc == 2 && res_arg == 1) stream_params.port = atoi(argv[1]);
    if (argc > 2) stream_params.port = atoi(argv[2]);

    returned_state = zed.enableStreaming(stream_params);
    if (returned_state != ERROR_CODE::SUCCESS) {
        print("Streaming initialization error: ", returned_state);
        return EXIT_FAILURE;
    }

    print("Streaming on port " + to_string(stream_params.port));

    SetCtrlHandler();
    sl::SensorsData mSensorData;
    while (!exit_app) {
        if (zed.grab() != ERROR_CODE::SUCCESS) {
            sleep_ms(1);
        }else{
[... 396 lines cut ...]
            // Stream input mode - IP + port
            string ip_adress = to_string(a) + "." + to_string(b) + "." + to_string(c) + "." + to_string(d);
            param.input.setFromStream(sl::String(ip_adress.c_str()), port);
            cout << "[Sample] Using Stream input, IP : " << ip_adress << ", port : " << port << endl;
        } else if (sscanf(arg.c_str(), "%u.%u.%u.%u", &a, &b, &c, &d) == 4) {
            // Stream input mode - IP only
            param.input.setFromStream(sl::String(argv[1]));
            cout << "[Sample] Using Stream input, IP : " << argv[1] << endl;
        } else if (arg.find("HD2K") != string::npos) {
            param.camera_resolution = sl::RESOLUTION::HD2K;
            cout << "[Sample] Using Camera in resolution HD2K" << endl;
        } else if (arg.find("HD1080") != string::npos) {
            param.camera_resolution = sl::RESOLUTION::HD1080;
            cout << "[Sample] Using Camera in resolution HD1080" << endl;
        } else if (arg.find("HD720") != string::npos) {
            param.camera_resolution = sl::RESOLUTION::HD720;
            cout << "[Sample] Using Camera in resolution HD720" << endl;
        } else if (arg.find("VGA") != string::npos) {
            param.camera_resolution = sl::RESOLUTION::VGA;
            cout << "[Sample] Using Camera in resolution VGA" << endl;
        }
    } else {
        // Default
    }
}

obraun-sl, 2021-01-19:
Hi,
You are right, it should be the same. (as well as timestamp which are also different).
This is probably a bug inside the SDK. We will look at it and provide a fix in next patched release (3.4.1)

obraun-sl, 2021-01-27:
Fix has been pushed  in the 3.4.0 available here : 
https://www.stereolabs.com/developers/release/

phamdat09, 2021-01-27:
Thanks for your support !!

closed 2021-01-27 by phamdat09