stereolabs/zed-sdk #338: ZED Calibration file not found when run as startup service

birdistheword96, 2021-01-20:
I have an application which uses the ZED2 camera for depth measurement, and when I login and run the script as the main user everything works fine. However, the end goal is for this application to run headless, so to negate the user needing to connect a keyboard to login and then navigate to the application and run it, this has been made into a system service called 'detectionystem' that runs a script called `launch.sh` containing the instructions to start the application, which will start on bootup (when enabled with `sudo systemctl enable detectionsystem`).

However, whenever I try to run it with `sudo systemctl start detectionsystem` it doesn't work, and the output of `sudo systemctl status detectionsystem` is:

```
Jan 11 11:08:47 dt-desktop launch.sh[10231]: filename: /usr/local/zed/settings/SN24806336.conf is open: 0
Jan 11 11:08:47 dt-desktop launch.sh[10231]: [ZED][Init] Unable to download calibration file.
Jan 11 11:08:47 dt-desktop launch.sh[10231]:  * Option 1: Launch ZED Explorer App with --dc 24806336
Jan 11 11:08:47 dt-desktop launch.sh[10231]:  * Option 2: Download your calibration file on https://calib.stereolabs.com
Jan 11 11:08:47 dt-desktop launch.sh[10231]: CALIBRATION FILE NOT AVAILABLE
Jan 11 11:08:47 dt-desktop launch.sh[10231]: [ZED]ERROR: sl::Camera::Open has not been called, no Camera instance running.
```

I know the file exists as I can navigate to it if I add `ls /usr/local/zed/settings/` to the beginning of the startup script, and I have ensured that it has full permissions (`ls -la` lists `-rwxrwxrwx`).

There are no other applications running that are using the camera at the time, so what could be causing this issue?

adujardin, 2021-11-30:
Could you post the calibration file content here?

Could this be that the service starts before the network, then the SDK tries to download the calibration file and can't? Or simply are you sure that there's internet access?

I should mention that the calibration file download only happened once at the first startup, and can also be downloaded manually if needed (I understand that it's not compatible with your use case, but it's a workaround).

closed 2022-04-26 by github-actions[bot]

ebrukilic61, 2023-02-14:
I have the same issue, when I try to run my code, zed camera gives me the error -> "CALIBRATION FILE NOT AVAILABLE" and "CAMERA STREAM FAILED TO START". What should I do to fix this error? Can anyone help me?