stereolabs/zed-sdk #342: ModuleNotFoundError: No module named 'ogl_viewer'

AdeQQ, 2021-02-21:
Hi! 
When I'm trying to run the python object detection example in vs code I get the following:
`Traceback (most recent call last):
  File "e:\zed_python_object_detection.py", line 26, in <module>
    import ogl_viewer.viewer as gl
ModuleNotFoundError: No module named 'ogl_viewer'`.
I have all the packages installed and up to date. Anybody had such an issue?

closed 2021-02-21 by AdeQQ

accolypse1994, 2021-03-04:
Hey I have the same issue. Were you able to solve it ?

qt-truong, 2021-03-04:
Hi @accolypse1994,

You can check https://github.com/stereolabs/zed-examples/issues/319#issuecomment-765250638 which could be related to your problem

cgsuccesscg, 2022-09-07:
Hey I have the same issue. Were you able to solve it ?