stereolabs/zed-sdk #343: Retrieving Camera Pose

Zumbalamambo, 2021-02-21:
I'm using the following code to retrieve the camera pose (4x4 Homogeneous matrix with respect to the real world).

```
import sys
import time

import cv2
import pyzed.sl as sl
import ogl_viewer.viewer as gl

def transform_pose(pose, tx):
    transform_ = sl.Transform()
    transform_.set_identity()
    # Translate the tracking frame by tx along the X axis
    transform_[0][3] = tx
    # Pose(new reference frame) = M.inverse() * pose (camera frame) * M, where M is the transform between the two frames
    transform_inv = sl.Transform()
    transform_inv.init_matrix(transform_)
    transform_inv.inverse()
    pose = transform_inv * pose * transform_

    return pose

def main():
    print("Running Spatial Mapping sample ... Press 'q' to quit")
    # Create a Camera object
    zed = sl.Camera()
    image = sl.Mat()
    depth = sl.Mat()
    # Create a InitParameters object and set configuration parameters
    init_params = sl.InitParameters()
    init_params.camera_resolution = sl.RESOLUTION.HD720  # Use HD720 video mode
    init_params.coordinate_units = sl.UNIT.METER  # Set coordinate units
    init_params.coordinate_system = sl.COORDINATE_SYSTEM.RIGHT_HANDED_Y_UP  # OpenGL coordinates

    # If applicable, use the SVO given as parameter
    # Otherwise use ZED live stream
    if len(sys.argv) == 2:
        filepath = sys.argv[1]
        print("Using SVO file: {0}".format(filepath))
        init_params.set_from_svo_file(filepath)

    # Open the camera
    status = zed.open(init_params)
    if status != sl.ERROR_CODE.SUCCESS:
        print(repr(status))
        exit()

    # Get camera parameters
    camera_parameters = zed.get_camera_information().camera_configuration.calibration_parameters.left_cam

    pymesh = sl.Mesh()  # Current incremental mesh
    image = sl.Mat()  # Left image from camera
    pose = sl.Pose()  # Camera pose tracking data
    camera_pose = sl.Pose()

    viewer = gl.GLViewer()
    viewer.init(camera_parameters, pymesh)

    spatial_mapping_parameters = sl.SpatialMappingParameters()
    tracking_state = sl.POSITIONAL_TRACKING_STATE.OFF
    mapping_state = sl.SPATIAL_MAPPING_STATE.NOT_ENABLED
    mapping_activated = False
    last_call = time.time()  # Timestamp of last mesh request

    # Enable positional tracking
    err = zed.enable_positional_tracking()
    if err != sl.ERROR_CODE.SUCCESS:
        print(repr(err))
        exit()

    tracking_parameters = sl.PositionalTrackingParameters()
    err = zed.enable_positional_tracking(tracking_parameters)
    py_translation = sl.Translation()

    # Set runtime parameters
    runtime = sl.RuntimeParameters()
    should_save = False
    while viewer.is_available():
        # Grab an image, a RuntimeParameters object must be given to grab()
        if zed.grab(runtime) == sl.ERROR_CODE.SUCCESS:
            # Retrieve left image
            zed.retrieve_image(image, sl.VIEW.LEFT)
            # Update pose data (used for projection of the mesh over the current image)
            tracking_state = zed.get_position(pose)

            if mapping_activated:
                mapping_state = zed.get_spatial_mapping_state()
                # Compute elapsed time since the last call of Camera.request_spatial_map_async()
                duration = time.time() - last_call
                # Ask for a mesh update if 500ms elapsed since last request
                if (duration > .5 and viewer.chunks_updated()):
                    zed.request_spatial_map_async()
                    last_call = time.time()

                if zed.get_spatial_map_request_status_async() == sl.ERROR_CODE.SUCCESS:
                    zed.retrieve_spatial_map_async(pymesh)
                    viewer.update_chunks()

            change_state = viewer.update_view(image, pose.pose_data(), tracking_state, mapping_state)
            if change_state:
                print("Pressed Space Bar...")
                if not mapping_activated:
                    should_save = True
                    print("Resetting Transform...")
                    init_pose = sl.Transform()
                    zed.reset_positional_tracking(init_pose)
                    mapping_parameters = sl.SpatialMappingParameters()
                    mapping_parameters.resolution_meter = mapping_parameters.get_resolution_preset(
                        sl.MAPPING_RESOLUTION.LOW)  # Or use preset
                    mapping_parameters.range_meter = mapping_parameters.get_range_preset(sl.MAPPING_RANGE.MEDIUM)
                    mapping_parameters.map_type = sl.SPATIAL_MAP_TYPE.MESH
                    mapping_parameters.save_texture = True  # Scene texture will be recorded
                    mapping_parameters.set_resolution(sl.MAPPING_RESOLUTION.MEDIUM)
                    filter_params = sl.MeshFilterParameters()  # not available for fused point cloud
                    filter_params.set(sl.MESH_FILTER.LOW)  # not available for fused point cloud

                    # Enable tracking and mapping
                    tracking_parameters = sl.PositionalTrackingParameters()
                    zed.enable_positional_tracking(tracking_parameters)
                    zed.enable_spatial_mapping(mapping_parameters)
                    # Clear previous mesh data
                    pymesh.clear()
                    viewer.clear_current_mesh()

                    # Start timer
                    last_call = time.time()

                    mapping_activated = True
                else:

                    # Extract whole mesh
                    zed.extract_whole_spatial_map(pymesh)

                    filter_params = sl.MeshFilterParameters()
                    filter_params.set(sl.MESH_FILTER.MEDIUM)
                    # Filter the extracted mesh
                    pymesh.filter(filter_params, True)
                    pymesh.apply_texture()
                    viewer.clear_current_mesh()

                    # If textures have been saved during spatial mapping, apply them to the mesh
                    if spatial_mapping_parameters.save_texture:
                        print("Save texture set to : {}".format(spatial_mapping_parameters.save_texture))
                        pymesh.apply_texture(sl.MESH_TEXTURE_FORMAT.RGBA)

                    # Save mesh as an obj file
                    filepath = "assets/mesh.obj"
                    status = pymesh.save(filepath)
                    if status:
                        print("Mesh saved under " + filepath)
                    else:
                        print("Failed to save the mesh under " + filepath)

                    mapping_state = sl.SPATIAL_MAPPING_STATE.NOT_ENABLED
                    mapping_activated = False
            if should_save:
                zed.retrieve_image(image, sl.VIEW.LEFT)
                zed.retrieve_measure(depth, sl.MEASURE.DEPTH)

                tracking_state = zed.get_position(camera_pose)
                image_ocv = image.get_data()
                depth_image_ocv = depth.get_data()

                if tracking_state == sl.POSITIONAL_TRACKING_STATE.OK:
                    timestamp = zed.get_timestamp(sl.TIME_REFERENCE.IMAGE).get_milliseconds()  # Get the image timestamp
                    translation_left_to_center = zed.get_camera_information().calibration_parameters.T[0]
                    pos_data = camera_pose.pose_data(sl.Transform())
                    pose = transform_pose(pos_data, translation_left_to_center)
                    save_depth = (depth.write("assets/depth/" + str(timestamp) + ".png") == sl.ERROR_CODE.SUCCESS)
                    save_image = (image.write("assets/rgb/" + str(timestamp) + ".jpg") == sl.ERROR_CODE.SUCCESS)

    print("Pressed Q")
    image.free(memory_type=sl.MEM.CPU)
    pymesh.clear()
    # Disable modules and close camera
    zed.disable_spatial_mapping()
    zed.disable_positional_tracking()
    zed.close()

if __name__ == "__main__":
    main()
```

The problem is that it throws the following error message,
```
transform_[0][3] = tx
  File "pyzed/sl.pyx", line 754, in pyzed.sl.Matrix4f.__getitem__
TypeError: 'int' object is not subscriptable
```

How can I retrieve the proper 4x4 matrix which defines the camera pose?

qt-truong, 2021-02-22:
Hi @Zumbalamambo,

You actually retrieved the camera pose when calling `tracking_state = zed.get_position(camera_pose)`. The pose of the camera is contained in `camera_pose`, of type [sl.Pose](https://www.stereolabs.com/docs/api/python/classpyzed_1_1sl_1_1Pose.html).
Then you extracted the [sl.Transform](https://www.stereolabs.com/docs/api/python/classpyzed_1_1sl_1_1Pose.html#a84c6fa0d5954ea1cbb8b1efe29ae9e39) object from the pose by calling `pos_data = camera_pose.pose_data(sl.Transform())`.

Also, if you want to set a Matrix4f element to a certain value, you should do instead : `transform_[0,3] = tx`

Hope it helps

Zumbalamambo, 2021-02-22:
@qt-truong As you may see from my code, I have set `transform_[0,3] = tx` in the method `transform_pose` but then it throws `TypeError: 'int' object is not subscriptable`

Zumbalamambo, 2021-02-22:
@qt-truong when I tried to print `pos_data`, it prints the following,
```

55A4E4058810
0.999938 -0.011140 0.000185 0.000000
0.011105 0.997873 0.064237 0.000000
-0.000900 -0.064230 0.997935 0.000000
0.000000 0.000000 0.000000 1.000000
```

what exactly is the number 55A4E4058810?

qt-truong, 2021-02-23:
Hi @Zumbalamambo,

You actually called `transform_[0][3] = tx` which is why you had this error. You need to replace the double subscripts by `transform_[0,3] = tx`.

As for `pos_data`, you don't need to worry about the first characters. Your sl.Transform object will contain the matrix data that is printed after that. For more information on sl.Transform objects and what information it contains, you can check the API reference page [here](https://www.stereolabs.com/docs/api/python/classpyzed_1_1sl_1_1Transform.html)

Zumbalamambo, 2021-02-23:
1. I have made those changes and now it throws the following error in the final line,

```
 pose = transform_inv * pose * transform_
  File "pyzed/sl.pyx", line 742, in pyzed.sl.Matrix4f.__mul__
TypeError: Argument must be Matrix4f or scalar type.
```

2. I'm trying to ignore the first few alpha numric character only serialize only the 4x4 matrix but then I don't find any API document which defines how I could remove the first few character... It is always attached to the pose_data :(

qt-truong, 2021-02-23:
1. As the error log suggests, the operation you are trying to perform is only valid on scalar or Matrix4f objects. Here, your objects are of type sl.Transform. In order to have a valid operation, you need to access the content of your Transform object by doing :
`pose = transform_inv.m * pose.m * transform_.m`

2. Similarly, calling `pos_data.m` will output the content of your Transform in a numpy array. Maybe that is what you're looking for ? (see [here](https://www.stereolabs.com/docs/api/python/classpyzed_1_1sl_1_1Matrix4f.html#a3355f83c4087eea9b05d7778b5cdb52d) for documentation)

closed 2021-02-23 by Zumbalamambo