stereolabs/zed-sdk #345: Using Anaconda - Spyder IDE to run tutorial

vincentsiowyongren, 2021-02-24:
As Anaconda contains all the pre-requisite packages such as cython and numpy, I wanted to run the tutorial - Hello ZED on it to try. However, there are no visible error except for the SystemExit: 1 when i run it. Anyone tried running it using anaconda?

qt-truong, 2021-02-24:
Hi @Vincentyr,

Have you installed the Python API inside your virtual environment ? If that is the case, please also make sure that the camera is well plugged in and detected by your system

Vincentyr, 2021-02-24:
Yes, i am using anaconda and i followed the steps in the website - https://www.stereolabs.com/docs/app-development/python/virtual_env/. I was able to return my serial number but the camera does not open 

This is the output for reference : runfile('C:/Users/Yong Ren/Documents/Python Scripts/Hello ZED.py', wdir='C:/Users/Yong Ren/Documents/Python Scripts')
Hello! This is my serial number: XXXX

P-yver, 2021-02-25:
The 1st tutorial only prints the camera serial number, just after [open](https://github.com/stereolabs/zed-examples/blob/master/tutorials/tutorial%201%20-%20hello%20ZED/python/hello_zed.py#L33).
It looks like everything is fine if your output is:
`Hello! This is my serial number: XXXX`
Simply try the next tutorial or a sample.

closed 2022-04-26 by github-actions[bot]