stereolabs/zed-sdk #346: Get 3d point cloud manually issues

afloralblb, 2021-02-25:
Hi,

I have two questions:

1. I am trying to use the depth map as an input of some algorithms and use the output depth map to get 3d point cloud, so i need to calculate the 3d point cloud manually. I checked 'SN****.conf' file, i only found cx,cy,fx,fy but i cannot find the scaling factor, is it contained in 'SN****.conf' file?

2. Currently I got 8-bit depth map which scales in [0,255], is it possible to using such kind of depth map to get 3d point cloud manually or i have to use depth_map.getValue(x,y) to get raw depth value to do calculation?

Thanks

obraun-sl, 2021-02-25:
Hi,

1. You already have anything you need. To convert a pixel (u,v) into a 3D point, you just need the depth and the camera calibration matrix (cx,cy,fx,fy): 
here is the formulae
float z = depth.getValue(u,v);
P3D.x = (u - cx) * z / fx;
P3D.y = (v - cy) * z / fy;
P3D.z = z;

2. You cannot use a 8bits representation of the depth map. It does not have any metrics inside. Use retrieveMeasure instead of retrieveImage.
See here : 
https://github.com/stereolabs/zed-examples/blob/master/tutorials/tutorial%203%20-%20depth%20sensing/cpp/main.cpp#L58

closed 2021-03-12 by obraun-sl

an99990, 2021-10-15:
are the camera intrinsic suppose to change per frame ? i got 0 for all four values using self.zed.get_camera_information().calibration_parameters.left_cam.

an99990, 2021-10-18:
issue was that i opened the camera after getting intrinsic values. you have to open zed then get camera info.