stereolabs/zed-sdk #347: frame rate problem

SimonSongg, 2021-03-04:
I used Jetson Xavier NX, ubuntu 18, cuda 10.2 and ZED 2. Using depth viewer, 720p, standard mode, even with performance mode, the frame rate is like 35hz. But from a youtube video, someone used TX2 and zed depth viewer and got 40hz. TX2 is weaker than Xavier NX, I don't know why I can't get higher frequency that he did. The screenshot of the video is attached below.
[image]
[image]

obraun-sl, 2021-03-05:
Is your Xaiver NX setup in MAXN performance mode ? (using nvpmodel)

SimonSongg, 2021-03-05:
> Is your Xaiver NX setup in MAXN performance mode ? (using nvpmodel)

Yes, 15W, 6core

SimonSongg, 2021-03-05:
> Is your Xaiver NX setup in MAXN performance mode ? (using nvpmodel)

Did you test it on Xavier NX? I think I have had set the max power mode, I tried15w 2core (1.1Ghz GPU and 1.9 Ghz CPU) and 15w 6core (1.1Ghz GPU and 1.4Ghz CPU (6core)), no performance improvements.

closed 2022-04-26 by github-actions[bot]