stereolabs/zed-sdk #348: Saving depth image issue

Yakamoko, 2021-03-05:
Hi there!

Im working on a project and im taking both rgb and depth images from zed mini camera.

I have the 3.3.1 sdk version on cuda 10.2 win10. My problem is that i cannot read the depth image correct since the mat type is F32C1 and unfortunately no image format (png, tiff, pfm) can be read with F32C1 on openCV.

How am i going to solve that problem? I checked that in your 3.4.1 version sdk you support U16_C1 as mat type format while in the sdk version im using its not. Although in your docs u claim depth mat  type Format as F32C1 do you think if i stored it on U16_C1 like all depth images would it be a problem?

Thanks in advance !

obraun-sl, 2021-03-05:
Hi,
You can use sl::Mat::write() function even with a Mat of type 32FC1. Behind, it will convert to 16UC1 for writting the PNG file.
Just make sure that the InitParameter::units is in millimeter to avoid data loss.

Yakamoko, 2021-03-06:
Thank you so much for your answer obraun!

Im gonna check it out right now

Greetings,
Yakamoko

closed 2021-03-12 by obraun-sl